node-opcua/node-opcua #951: Documentation: MessageSecurityMode link is misleading

adrfantini, 2021-02-18:
Hi, I was reading the documentation and found it quite confusing when I could not find the possible values of the parameter SecurityMode: https://node-opcua.github.io/api_doc/2.20.0/interfaces/opcuaclientbaseoptions.html#securitymode

The doc links to here: https://node-opcua.github.io/api_doc/2.20.0/enums/datatypeids.html#messagesecuritymode
Which only says `MessageSecurityMode: = 302`. What does it mean?

Only manually searching I found out the specific page for this enum:
https://node-opcua.github.io/api_doc/2.20.0/enums/messagesecuritymode.html

Which correctly and clearly states the possible values. I'm unsure whether this is the expected documentation organisation, or if this is a documentation bug.

Graos, 2021-02-18:
MessageSecurityMode: = 302 is the OPC UA  integer ID defined for that node in the OPC UA specification. For the values you need to refer the below enum values
Enumeration members
Invalid
None
Sign
SignAndEncrypt

adrfantini, 2021-02-18:
Thanks for the reply.I understand this. But shouldn't the documentation point to the enum valuess, and not o the node ID?

erossignon, 2021-03-02:
Thank you for pointing this out. The documentation has been regenerated : 
http://node-opcua.github.io/api_doc/2.32.0/index.html
https://node-opcua.github.io/api_doc/2.32.0/interfaces/opcuaclientbaseoptions.html#securitymode

closed 2021-03-02 by erossignon