node-opcua/node-opcua #952: Extract datatype from nodeId

icemini, 2021-02-26:
Is it possible to extract the datatype of a node from a server? I'm browsing through my nodes folder and want to extract more information regarding the datatype of the node by providing the nodeid. I've tried using this getBuiltInDataType function:

`try {
                       const nodeId = await the_session.read({
                           nodeId: opcua.resolveNodeId(val.nodeId),
                       });
                       const res = await the_session.getBuiltInDataType(nodeId.value.value);
                   } catch (e) {
                       console.log('Catch an error: ', e)
                   }`

But end up getting an Error:

"Error: cannot read DataType Attribute BadAttributeIdInvalid (0x80350000)"

Is there something I'm doing wrong?

Thank you

AndreasHeine, 2021-02-26:
```js
    const readResult = await session.read({nodeId: "i=2267", attributeId: opcua.AttributeIds.DataType});
    console.log(readResult);
```

Output:
> DataValue {
>   value: Variant {
>     dataType: 17,
>     arrayType: 0,
>     value: NodeId { identifierType: 1, value: 3, namespace: 0 },
>     dimensions: null
>   },
>   statusCode: ConstantStatusCode {
>     _value: 0,
>     _description: 'No Error',
>     _name: 'Good'
>   },
>   sourceTimestamp: null,
>   sourcePicoseconds: 0,
>   serverTimestamp: 2021-02-26T18:32:16.786Z {
>     high_low: [ 30870637, -1248790943 ],
>     picoseconds: 390500000
>   },
>   serverPicoseconds: 390500000
> }

NodeId { identifierType: 1, value: 3, namespace: 0 }
https://raw.githubusercontent.com/OPCFoundation/UA-Nodeset/v1.04/Schema/NodeIds.csv
--> ObjectIds 3 --> byte

icemini, 2021-03-01:
Thank you for the answer!

Unfortunately I still get the "Error: cannot read DataType Attribute BadAttributeIdInvalid (0x80350000)" error when using the getBuiltInDatatype funktion

AndreasHeine, 2021-03-01:
http://node-opcua.github.io/api_doc/2.32.0/interfaces/node_opcua.clientsession.html#getbuiltindatatype

is `nodeId.value.value` really a nodeid? it does not appear to! 
BadAttributeIdInvalid: typically tells you that the thing you want to get the type of does not have a this attribute

can you make some console.log(nodeId) and console.log(nodeId.value.value) and tell/post what it actually is?

icemini, 2021-03-01:
The nodeId.value.value prints a NodeId object with the following properties:

 NodeId {
  identifierType: ....,
  value: ...,
  namespace: ...
}

AndreasHeine, 2021-03-01:
does the nodeid which got printed have a DataType Attribute?

you should test it against a standard server node (only variables and properties have datatypes) like "ns=0;i=2253"
[image]

session.getBuildInDataType(nodeId) --> should return a number in case of "ns=0;i=2253" -->12 --> String

icemini, 2021-03-01:
The whole nodeId that is retrieved from session read is:

DataValue {
  value: Variant {
    dataType: 17,
    arrayType: 0,
    value: NodeId {
      identifierType: 2,
      value: '..',
      namespace: 4
    },
    dimensions: null
  },
  statusCode: ConstantStatusCode {
    _value: 0,
    _description: 'No Error',
    _name: 'Good'
  },
  sourceTimestamp: null,
  sourcePicoseconds: 0,
  serverTimestamp: 2021-03-01T12:02:50.532Z {
    high_low: [ 30871186, -848380352 ],
    picoseconds: 0
  },
  serverPicoseconds: 0
}

So it does seem to have a datatype, however I'm not sure I'm passing it correctly to the function. Maybe getBuildInDataType is the wrong function to use if I want more info regarding the type

AndreasHeine, 2021-03-01:
```js
const readDataValueResult = await the_session.read({
                           nodeId: opcua.resolveNodeId("ns=0;i=2253"),
                       });
```

`readDataValueResult.value.value.value` should hold a number which is the equivalent to the "DataType Enumeration"
http://node-opcua.github.io/api_doc/2.32.0/enums/node_opcua.datatype.html

if you want to use "getBuildInDataType" you should call:
```js
const dtype = await the_session.getBuiltInDataType(opcua.coerceNodeId("ns=0;i=2253"));
```

if you say type do you mean:
DataType of Value (Variable Value="Test" -> you want to figure out if it is a string) -> read or getBuiltInDataType
or 
TypeDefinition of the Node -> browse (get the TypeDefinition reference of a node like is it a BasObjectType or a FolderType or ...)

erossignon, 2021-03-02:
duplicates #586

closed 2022-02-27 by erossignon