node-opcua/node-opcua #956: Connection, re-connection without event based and read Tag values without event based

ganesh1306, 2021-03-12:
Hi Thanks for providing node-opcua as open source
i want connect to server and read data  with-out event based that has to in while loop 
means once it got connected to server i'll set the flag if got disconnected i'll disable flag 
if connection successful i need to read the values in while loop if connection lost then try for connection if server up again then read the values 

this is the sample code i used with event based

	let endpointUrl = ini_parse.opcConfig.url
		client = OPCUAClient.create({

			keepSessionAlive: true,
			requestedSessionTimeout: 60000,
			securityMode: mode,
			securityPolicy:policy,
			requestedSessionTimeout: 10000,
			applicationName: "NodeOPCUA-Client",
			certificateFile: certpath,
			privateKeyFile:keypath,
			endpoint_must_exist: false,

			connectionStrategy: {
				maxRetry: 10000000,
				initialDelay: 100,
				maxDelay: 1000
			},
			
		});

		await client.connect(endpointUrl, function (err) {
			if (err)
				console.log(" cannot connect to endpoint :", endpointUrl);
			else {
				console.log("connected !");                
			}
		})

		 
		client.on("backoff", (retry, delay) => {
			console.log("Backoff ", retry, " next attempt in ", delay, "ms");
		});

		client.on("connection_lost", function (err) {
			if (err)
				console.log("Connection lost");
		});

		client.on("connection_reestablished", function (err) {
			if (err)
				console.log("Re-established Faield with server");
			else
				console.log(" connection Re-established ");

		});

		client.on("connection_failed", function  (err) {
			if(err)
				console.log("Connection failed");
		});

		client.on("start_reconnection", function (err) {
			if(err)
				console.log("Starting Failed reconnection");

			else
				console.log("Starting reconnection");
		});

		client.on("after_reconnection", (err) => {
			if (err)
				console.log("Failed After Reconnection event =>", err);
			else
				console.log("After Reconnection event ");
		});

		
		client.on("connection_failed", (err) => {
			console.log("Conneciton with server is Failed ")
		});

		client.on("reconnection_attempt_has_failed", function (err) {
			if(err)
			console.log("Reconnection Attemp failed Server shutdown")
		})
	}

this sample code i tried but it was not successful when i use event based for connection and read tag values it will push in the order how many valued i'm read like 
ns=0;i=2256
ns=0;i=2257
ns=0;i=15954
if i try to read the above tags it will be on event based after read the tag values if try to make json packet i'll get 3 times data at mqtt 

please can anyone suggest me on this

	while(1)
	{
		if(connectionwithserver())
		{
			//connect successfull
			//read the values 
			// after read the tag values push to mqtt
		}
		else
		{
			//re-try for connection
		}

	}

	function connectionwithserver()
	{
	try 
	{
	    //connect to server
		return true 
	}
	catch(err)
	{
		//failed to connect
		return false
	}
	function read_tagvalues()
	{
		//read the tag values from 10 -50 
		
	}

erossignon, 2021-03-17:
You're mixing async/await and callback... Please prefer the async syntax :

````
try {

   await client.connect(endpointUrl);
   console.log("connected !");
} catch(err) {
   console.log("cannot connect");
}                
```` 
You probably need to use subscription to get notification  when the value changes on the server side;

Please refer to sample 
https://github.com/node-opcua/node-opcua/blob/master/documentation/sample_client_ts.ts#L115
    
or alternatively https://leanpub.com/node-opcuabyexample

ganesh1306, 2021-04-14:
I have gone through the below link and section

"Setting up X509 extension on certificate" I couldn't able to find the cert.cnf file.

https://github.com/node-opcua/node-opcua/blob/master/documentation/notes_on_certificates.md

Is their any steps for certificate generation

ganesh1306, 2021-04-15:
I have gone through the node-opcua-pki i didn't find the certificate generation 

I installed the https://npm.io/package/node-opcua-pki when I run the command  in command-line I got Error 

'crypto_create_CA' is not recognized as an internal or external command,

operable program or batch file.

closed 2022-04-27 by stale[bot]