node-opcua/node-opcua #958: Siemen S7 1500 callback has thrown en error  Error: publishEngine: invalid state

erikholleboom, 2021-03-14:
**Current behavior**
Normal operation is no problem.
But when the PLC stops for a moment to update code or a variable the error below appears.
In this situation I have 2 PLC's connected.
PLC1 subscribes to about 1300 items by using 2 subscriptions.
PLC2 subscribed to about 600 items by a single subscription.

The problem only appears on PLC1. 
Order of connecting does not seem to have an effect.
The amount of subscriptions seems to have an effect. With PLC1 having only a single subscription (1000 elements) is works like a charm. Workaround having 2 connections with a single subscriptions seems to work fine as well.

**Describe the bug**
ZDS_ApplicatieSrv | ERROR !!! callback has thrown en error  Error: publishEngine: invalid state
ZDS_ApplicatieSrv |     at Object.assert (c:\gbs\storage\EcgLyzjHegx7nEPmj\bundle\programs\server\npm\node_modules\meteor\lbsgui-opcua\node_modules\node-opcua-assert\dist\index.js:13:21)
ZDS_ApplicatieSrv |     at ClientSidePublishEngine.suspend (c:\gbs\storage\EcgLyzjHegx7nEPmj\bundle\programs\server\npm\node_modules\meteor\lbsgui-opcua\node_modules\node-opcua-client\dist\private\client_publish_engine.js:54:29)
ZDS_ApplicatieSrv |     at suspend_old_session_publish_engine (c:\gbs\storage\EcgLyzjHegx7nEPmj\bundle\programs\server\npm\node_modules\meteor\lbsgui-opcua\node_modules\node-opcua-client\dist\reconnection.js:142:40)
ZDS_ApplicatieSrv |     at c:\gbs\storage\EcgLyzjHegx7nEPmj\bundle\programs\server\npm\node_modules\meteor\lbsgui-opcua\node_modules\async\dist\async.js:2948:28
ZDS_ApplicatieSrv |     at replenish (c:\gbs\storage\EcgLyzjHegx7nEPmj\bundle\programs\server\npm\node_modules\meteor\lbsgui-opcua\node_modules\async\dist\async.js:440:21)
ZDS_ApplicatieSrv |     at c:\gbs\storage\EcgLyzjHegx7nEPmj\bundle\programs\server\npm\node_modules\meteor\lbsgui-opcua\node_modules\async\dist\async.js:445:13
ZDS_ApplicatieSrv |     at eachOfLimit$1 (c:\gbs\storage\EcgLyzjHegx7nEPmj\bundle\programs\server\npm\node_modules\meteor\lbsgui-opcua\node_modules\async\dist\async.js:471:34)
ZDS_ApplicatieSrv |     at awaitable (c:\gbs\storage\EcgLyzjHegx7nEPmj\bundle\programs\server\npm\node_modules\meteor\lbsgui-opcua\node_modules\async\dist\async.js:208:32)
ZDS_ApplicatieSrv |     at eachOfSeries (c:\gbs\storage\EcgLyzjHegx7nEPmj\bundle\programs\server\npm\node_modules\meteor\lbsgui-opcua\node_modules\async\dist\async.js:658:16)
ZDS_ApplicatieSrv |     at awaitable (c:\gbs\storage\EcgLyzjHegx7nEPmj\bundle\programs\server\npm\node_modules\meteor\lbsgui-opcua\node_modules\async\dist\async.js:208:32)
ZDS_ApplicatieSrv |     at c:\gbs\storage\EcgLyzjHegx7nEPmj\bundle\programs\server\npm\node_modules\meteor\lbsgui-opcua\node_modules\async\dist\async.js:2947:9
ZDS_ApplicatieSrv |     at awaitable (c:\gbs\storage\EcgLyzjHegx7nEPmj\bundle\programs\server\npm\node_modules\meteor\lbsgui-opcua\node_modules\async\dist\async.js:208:32)
ZDS_ApplicatieSrv |     at Object.series (c:\gbs\storage\EcgLyzjHegx7nEPmj\bundle\programs\server\npm\node_modules\meteor\lbsgui-opcua\node_modules\async\dist\async.js:3957:16)
ZDS_ApplicatieSrv |     at repair_client_session_by_recreating_a_new_session (c:\gbs\storage\EcgLyzjHegx7nEPmj\bundle\programs\server\npm\node_modules\meteor\lbsgui-opcua\node_modules\node-opcua-client\dist\reconnection.js:136:11)
ZDS_ApplicatieSrv |     at c:\gbs\storage\EcgLyzjHegx7nEPmj\bundle\programs\server\npm\node_modules\meteor\lbsgui-opcua\node_modules\node-opcua-client\dist\reconnection.js:118:20
ZDS_ApplicatieSrv |     at wrapper (c:\gbs\storage\EcgLyzjHegx7nEPmj\bundle\programs\server\npm\node_modules\meteor\lbsgui-opcua\node_modules\async\dist\async.js:268:20)

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. Start nodeopc-ua with subscriptions to 2 PLC's
2. Make a code change in Tiaportal and update the PLC
3. the problem appears.

**Expected behavior**
Having a single session with multiple subscriptions not causing a pusblish out of sync error.

**Screenshots**
n.a.

**Gist**

No small way to reproduce

**Context**

- (x) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ____PC and PLC Siemen S7 1500__
- OS version: __Windows Server 2019____

    - ( ) Windows : version : _Windows Server 2019_____
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: _Siemens S7 1500_____
    - Version: ______
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: https://

- **node-opcua version:** :
  
  

- **Node:**
  `node --version` = 2.33.0

erossignon, 2021-03-17:
I suspect that PLC only supports one single subscription per session. However, node-opcua should handle this without raising an error.

AndreasHeine, 2021-03-17:
s7-1500: embedded 2017 opcua server profile should support more then one

erossignon, 2021-03-17:
The script fails inside a reconnection; 
```` 
odules\meteor\lbsgui-opcua\node_modules\node-opcua-client\dist\reconnection.js:136
```` 
It would be nice if you can provide a gist with a script that reproduces the issue. 
There could be some issue with the session or subscription parameters.
If you cannot share the code on a public forum, please contact sterfive directly for a private discussion.

mikakaraila, 2021-05-17:
I expect operation limit hits:
[image]

AndreasHeine, 2021-05-18:
@mikakaraila if the operational limits got hit the s7 throws BadTooManyOperations!

mikakaraila, 2021-05-18:
OK, I have now my own S7-1500 for testing. I will check this.

mikakaraila, 2021-05-18:
Now I got new error:
```javascript
2021-05-18T09:52:52.898Z: <INF> Interface Monitor: IPv4 network configuration changed
2021-05-18 12:52:52 error:  !!!!!!!!!!!!!!!!!!!!!!!!  Starting Reconnection !!!!!!!!!!!!!!!!!!!
 !!!!!!!!!!!!!!!!!!!!!!!!  Starting Reconnection !!!!!!!!!!!!!!!!!!! ClientName:Client4Siemens
2021-05-18 12:53:13 error: backoff  attempt #0 retrying in 5 seconds
backoff  attempt #0 retrying in 5 seconds
2021-05-18 12:53:22 info: DNA OPC UA Client uses memory ~142.72 MB
2021-05-18 12:53:22 info: DNA OPC UA Client change 1 MB
2021-05-18T09:53:22.225Z: <INF> Interface Monitor: IPv4 network configuration changed
2021-05-18T09:53:27.091Z: <INF> Interface Monitor: IPv4 network configuration changed
2021-05-18 12:53:40 error: backoff  attempt #1 retrying in 10 seconds
backoff  attempt #1 retrying in 10 seconds
2021-05-18 12:53:50 error:  !!!!!!!!!!!!!!!!!!!!!!!!  CONNECTION RE-ESTABLISHED !!!!!!!!!!!!!!!!!!!
 !!!!!!!!!!!!!!!!!!!!!!!!  CONNECTION RE-ESTABLISHED !!!!!!!!!!!!!!!!!!! ClientName:Client4Siemens
2021-05-18 12:53:50 error: Cannot read values: Error:  serviceResult = BadSessionNotActivated (0x80270000)
2021-05-18 12:53:50 error: watchDog error: Error:  serviceResult = BadSessionNotActivated (0x80270000)
2021-05-18 12:53:51 error: Cannot read values: Error:  serviceResult = BadSessionNotActivated (0x80270000)
2021-05-18 12:53:51 error: watchDog error: Error:  serviceResult = BadSessionNotActivated (0x80270000)
2021-05-18 12:53:52 error: Cannot read values: Error:  serviceResult = BadSessionNotActivated (0x80270000)
2021-05-18 12:53:52 error: watchDog error: Error:  serviceResult = BadSessionNotActivated (0x80270000)
```

mikakaraila, 2021-05-18:
Now I found something for you, after long break:

```javascript
2021-05-18 13:17:11 error:  !!!!!!!!!!!!!!!!!!!!!!!!  CONNECTION RE-ESTABLISHED !!!!!!!!!!!!!!!!!!!
 !!!!!!!!!!!!!!!!!!!!!!!!  CONNECTION RE-ESTABLISHED !!!!!!!!!!!!!!!!!!! ClientName:Client4Siemens
10:17:11.537Z :client_subscription_impl      :634    serviceResult = BadNothingToDo (0x800f0000)
(node:9296) UnhandledPromiseRejectionWarning: Error: publishEngine: invalid state
    at Object.assert (D:\GIT\OPCUA\mvp4-opcua\node_modules\node-opcua-assert\source\index.ts:14:21)
    at ClientSidePublishEngine.suspend (D:\GIT\OPCUA\mvp4-opcua\node_modules\node-opcua-client\source\private\client_publish_engine.ts:76:9)
    at suspend_old_session_publish_engine (D:\GIT\OPCUA\mvp4-opcua\node_modules\node-opcua-client\source\reconnection.ts:161:44)
    at D:\GIT\OPCUA\mvp4-opcua\node_modules\async\dist\async.js:2948:28
    at replenish (D:\GIT\OPCUA\mvp4-opcua\node_modules\async\dist\async.js:440:21)
    at D:\GIT\OPCUA\mvp4-opcua\node_modules\async\dist\async.js:445:13
    at eachOfLimit$1 (D:\GIT\OPCUA\mvp4-opcua\node_modules\async\dist\async.js:471:34)
    at awaitable (D:\GIT\OPCUA\mvp4-opcua\node_modules\async\dist\async.js:208:32)
    at eachOfSeries (D:\GIT\OPCUA\mvp4-opcua\node_modules\async\dist\async.js:658:16)
    at awaitable (D:\GIT\OPCUA\mvp4-opcua\node_modules\async\dist\async.js:208:32)
(node:9296) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 4)
(node:9296) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

erossignon, 2021-05-31:
this issue shall now be fixed in node-opcua@2.44.0, please confirm

erikholleboom, 2021-06-11:
> this issue shall now be fixed in ***@***.***, please confirm

Hi,

I've done some testing with the last version, no problem seen anymore.

So the problem can be closed imho.

Regards,
Erik

closed 2021-06-11 by erossignon