node-opcua/node-opcua #962: Invalid variant when trying to "setValueFromSource" on a Boolean Node

ebrithiljonas, 2021-03-16:
**Current behavior**

When you try to change the Value of an existing Node with the method "setValueFromSource" it gives the following exception, but only if you so it with a Boolean Value, with Double for example it works just fine.

```
Exception has occurred: Error: Invalid variant arrayType: Scalar  dataType: Boolean value:[object Object] (javascript type = object )
  at constructHook (D:\Projekte\Retrofit\retrofit\umati_server\node_modules\node-opcua-variant\source\variant.ts:462:19)
    at new Variant (D:\Projekte\Retrofit\retrofit\umati_server\node_modules\node-opcua-variant\source\variant.ts:111:26)
    at Function._coerceVariant [as coerce] (D:\Projekte\Retrofit\retrofit\umati_server\node_modules\node-opcua-variant\source\variant.ts:74:66)
    at UAVariable.setValueFromSource (D:\Projekte\Retrofit\retrofit\umati_server\node_modules\node-opcua-address-space\src\ua_variable.ts:642:27)
    at construct_my_address_space (D:\Projekte\Retrofit\retrofit\umati_server\src\index2.ts:50:22)
    at post_initialize (D:\Projekte\Retrofit\retrofit\umati_server\src\index2.ts:58:5)
    at Immediate._onImmediate (D:\Projekte\Retrofit\retrofit\umati_server\node_modules\node-opcua-server\source\opcua_server.ts:1120:25)
    at processImmediate (internal/timers.js:461:21)
    at process.topLevelDomainCallback (domain.js:144:15)
    at process.callbackTrampoline (internal/async_hooks.js:129:14)
```

**Describe the bug**

As far as I can see it, that there is no way of handling Booleans in the "coerceVariantType" method and therefore is doesn't convert the value to a proper type before using it.

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. download this code: https://gist.github.com/ebrithiljonas/421287f61bed24ef6aab530b1e02925b
2. run it

**Expected behavior**

It should create a Node with the type Boolean and with the defined function as a value.

**Gist**

 
gist: https://gist.github.com/ebrithiljonas/421287f61bed24ef6aab530b1e02925b

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (x) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: PC x86 64bit
- OS version: Windows 10 64 bit

    - (x) Windows : version : 10.0.18363 Build 18363
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- **node-opcua version:** : 2.36.0
  
  

- **Node:**
  `node --version` = 14.15.5

erossignon, 2021-03-17:
the setVariantFromSource parameter is incorrect in your gist.

``` javascript

// Wrong
variableNode.setValueFromSource({
            dataType: opcua.DataType.Double,
            value: {
                get:  () => new opcua.Variant({dataType: opcua.DataType.Double, value: 0.255 })
            }
        });
  
```

it should be:

``` javascript

// Correct
  variableNode.setValueFromSource({
            dataType: opcua.DataType.Double,
            value:  value: 0.255
   });
```

ebrithiljonas, 2021-03-17:
Sorry I didn't specified my use case correctly. I don't want to set them to a fixed value, I want them to return the value of a variable.

This new gist should represent it correctly: https://gist.github.com/ebrithiljonas/04a6c1e53986390547f0c237fba1a1d0
And these two variables will change overtime by some other part of my Software.

erossignon, 2021-03-18:
Just call setValueFromSource whenever the variable value change :
```` 
  variableNode.setValueFromSource({
            dataType: opcua.DataType.Booelan,
            value:  booleanVar 
   });
````

ebrithiljonas, 2021-03-18:
Ok this seems to work, thank you. But it is still weird that this is only necessary when using Booleans, with Doubles it works just fine the other way.

closed 2021-03-19 by erossignon