node-opcua/node-opcua #963: Unsubsribe by Subscription ID

reto88, 2021-03-18:
I am useing the opc ua node with socket.io. I have several connections with different subscription Id.
How can I` UASUBSCRIPTION.terminate()` by `SubscriptionId `when a connection is finished? At the moment it terminate all subscritpion.

erossignon, 2021-03-21:
If you wish to terminate a subscription, without closing a session, use 

```` 
    await subscription.terminate();
```` 

However, subscriptions will be automatically terminated when you close the session except if you set deleteSubscriptions to false

```` 
    await session.close(/* deleteSubscriptions */ false);
```` 

by default, you should be using

```` 
    await session.close();
```` 
which is equivalent to 
```` 
    await session.close(/* deleteSubscriptions */ true);
````

mikakaraila, 2021-03-21:
And for one variable similar way:
monitoredItem.terminate();

closed 2022-04-27 by stale[bot]