node-opcua/node-opcua #966: Just a question. Not an issue

RakshanPremsagar, 2021-03-20:
Well, this is a question and not an issue.
I am documenting about the feature supported by the stack. I would like to know, if this stack supports LDS-ME. I read one query being answered saying yes it does. However, I do not seem to find it concretely. Could you tell me if the stack supports LDS or LDSME on its own? 
I understand discovery services are implemented completely. Also, GDS is not done here right?

Many thanks,
Rakshan

erossignon, 2021-03-20:
Rakshan, yes LDS-DME is supported: 

 - node-opcua server's can be configured to register to a LDS-ME (Local Discovery Server with Multicast Extensions))
 - there is a full implementation of a Local Discovery Server with Multicast Extension:
    - you can run it this way 

```` 
npx node-opcua-local-discovery-server
```` 
 - GDS support exists in the open-source library:  user can implement their own GDS server or client. 
 - NodeOPCUA Server can be configured to provide the PushCertificate management support.

 - For more advanced usage, users can contact sterfive for advanced  GDS server build on node-opcua , which is currently not open source.

RakshanPremsagar, 2021-03-29:
Thank you very much for the detailed answer @erossignon

closed 2021-03-29 by RakshanPremsagar