node-opcua/node-opcua #967: Perhaps A Question? How to remove nodes?

mdridley, 2021-03-22:
I have bought the book, and searched it and Google'd, and while I can read and write node values, I cannot figure out how to delete nodes.

I found DeleteNodesRequest, but cannot figure out how to use it. Is there an example I can look at or is this part still a work in progress?

mikakaraila, 2021-03-22:
addressSpace.deleteNode(nodeId);

mikakaraila, 2021-03-22:
See SDK API doc:
http://node-opcua.github.io/api_doc/0.2.0/classes/AddressSpace.html#method_deleteNode

mdridley, 2021-03-22:
Can AddressSpace be used on the client side? Really looking to either remove from the client.

mikakaraila, 2021-03-23:
As addressSpace belongs to server you have to implement own method that you can call from the client.
I have implemented similar for our server.

mdridley, 2021-03-23:
So, that is not an option since the server is not node. 

@erossignon Is there anything else you would recommend here to ensure that I can clear out previous entries when writing new?

erossignon, 2021-03-27:
The node-management service is not implemented in node-opcua (eityher on server or on client)
Maybe because it has never been requested by our professional users and sponsors so far.

@mdridley  please be more descriptive here, we need to understand in which context your client would need to delete some node from the OPCUA address space of a remote server. can you describe the use case?

falottec, 2021-04-10:
Dear all,

I have a similar question(opcua-client side). How can I get a "handle" to a monitoredItem (e.g. based on NodeIds) from a group (or from a subscription) to be able to call .terminate() method on it.

The problem is, that I am using ClientMonitoredItemGroup.create to create monitoredItems as a group and I do not have access to the monitoredItem directly. 

Is there a way to achieve this, as it is not practcal to terminate a complete group, since some nodes may still be needed.

Thank you.

erossignon, 2021-05-08:
@falottec , monitored items in a group cannot easily remove individually. You will have to send a raw DeleteMonitoredItems request.

closed 2022-12-04 by erossignon