node-opcua/node-opcua #968: BadNodeIdUnknown  Siemens s7

jordicolom, 2021-03-22:
Hello,
i need help to read node variables and it's not posible:

whith session.browse
BrowseResult =  { /*BrowseResult*/
 statusCode                    /* StatusCode          */: Good (0x00000)
 continuationPoint             /* ByteString          */: null
 references                    /* ReferenceDescript[] */: [
   { /*0*/
     referenceTypeId           /* NodeId              */: HasTypeDefinition (ns=0;i=40)
     isForward                 /* UABoolean           */: true
     nodeId                    /* ExpandedNodeId      */: ns=0;i=63
     browseName                /* QualifiedName       */: BaseDataVariableType
     displayName               /* LocalizedText       */: locale=en text=BaseDataVariableType
     nodeClass                 /* NodeClass           */: NodeClass.ObjectType ( 8)
     typeDefinition            /* ExpandedNodeId      */: ns=0;i=0
   }
 ]
};

with read:

{ /* DataValue */
   value: Variant(Scalar<Null>, value: <null>)
   statusCode:      BadNodeIdUnknown (0x80340000)
   serverTimestamp: 2021-03-22T10:34:23.115Z $ 007.700.000
   sourceTimestamp: null
}

my code:

const {
    OPCUAClient,
    NodeClass,
    AttributeIds,
    ClientSubscription,
    TimestampsToReturn
} = require("node-opcua");

var nodeId = "ns=7;s=S7300ET200M station_1.PLC_1.REGISTROS_BOCAS.BOCA[1].Codigo_Barras"; 

const endpointUri = "opc.tcp://localhost:4845";

(async () => {
   
    const client = OPCUAClient.create({ endpoint_must_exist: false});
    client.on("backoff", () => console.log("Backoff: trying to connect to ", endpointUri));

    await client.withSessionAsync(endpointUri, async (session) => {
        
        let browseResult = await session.browse({
            nodeId,
            nodeClassMask: NodeClass.Variable, // we only want sub node that are Variables
            resultMask: 63 // extract all information possible 
        });
        console.log("BrowseResult = ", browseResult.toString());
      
      nodeId ="ns=7;s=\"S7300ET200M station_1.PLC_1.REGISTROS_BOCAS.BOCA[1].Codigo_Barras\"";
       const maxAge = 0;
        const nodeToRead = {
            nodeId:nodeId,
            attributeId: AttributeIds.Value
        };
        const dataValue = await session.read(nodeToRead, maxAge);
        console.log(dataValue.toString());

    });
})();

thanks

jonathanschneider, 2021-04-17:
Maybe these are dumb questions but it seems like you're trying to communicate with an S7-300 controller via OPC UA. The S7-300 doesn't support OPC UA. Secondly, why is your endpoint the localhost?

AndreasHeine, 2021-04-17:
@jonathanschneider thats true but there siemens gateway products out there which run on a normal ipc so localhost is fine! https://github.com/node-opcua/node-opcua/issues/970

i am currently testing some stuff and it looks like there might be a bug with extension objects (especially with plc udt's which are custom datatypes in the end) but this is just a suggestion

AndreasHeine, 2021-04-17:
@jordicolom 
What is the TypeDefinition of:
REGISTROS_BOCAS ?

jonathanschneider, 2021-04-17:
> @jonathanschneider thats true but there siemens gateway products out there which run on a normal ipc so localhost is fine! #970
>
> i am currently testing some stuff and it looks like there might be a [...]

I figured it’s very unlikely that this could be the problem and thus deleted my comment again.

closed 2022-04-27 by stale[bot]