node-opcua/node-opcua #969: How enable and disable subscription?

mariodiciccio, 2021-03-22:
Hi,
there is an example for this issue?

var subscription = new ClientSubscription(session, {
    requestedPublishingInterval: 100,
    requestedLifetimeCount: 10 ,
    requestedMaxKeepAliveCount: 100,
    maxNotificationsPerPublish: 1000,
    publishingEnabled: false,
    priority: 6
});

setTimeout(()=>{
  subscription.publishingEnabled = true
}, 3000);

Thanks

erossignon, 2021-03-22:
await subscription.setPublishingMode(true);

mariodiciccio, 2021-03-24:
Thank you very much, but the first call is a lot slow about 1 min, while the next calls is fast about less 1 sec, why?

erossignon, 2021-03-29:
I would investigate what's going on in the server, The client just sends a simple command here.

mariodiciccio, 2021-03-30:
How can I help you?

erossignon, 2021-04-05:
You could help you/me by subscribing to support (https://support.sterfive.com) so we can investigate deeper on your specific setup, as suggested in the issue template.

closed 2022-04-27 by stale[bot]