node-opcua/node-opcua #970: what is the way to read a value ?

jordicolom, 2021-03-24:
Hello, 

I need to read / write values with a tree in opcscout as follows: 
S7300ET200M station_1.PLC_1.REGISTROS_BOCAS.BOCA[1].Codigo_Barras

I use:

nodeId ="ns=7;s=\"S7300ET200M station_1.PLC_1.REGISTROS_BOCAS.BOCA[1].Codigo_Barras\"";
const maxAge = 0;
const nodeToRead = {
  nodeId:nodeId,
  attributeId: AttributeIds.Value
};
const dataValue = await session.read(nodeToRead, maxAge);
console.log(dataValue.toString());

And i have this error:

{ /* DataValue */
   value: Variant(Scalar<Null>, value: <null>)
   statusCode:      BadNodeIdUnknown (0x80340000)
   serverTimestamp: 2021-03-22T10:34:23.115Z $ 007.700.000
   sourceTimestamp: null
}

what is the way to read a value ?

i attch  an image from my tree in opcscout:
[image]

AndreasHeine, 2021-03-24:
nodeId =`"ns=7;s="S7300ET200M station_1.PLC_1.REGISTROS_BOCAS.BOCA[1].Codigo_Barras"`;

jordicolom, 2021-03-24:
Sorry, it was a mistake when writing here, I really use this, I have the same problem :

const maxAge = 0;
const nodeToRead = {
       nodeId: 'ns=7;s="S7300ET200M station_1.PLC_1.REGISTROS_BOCAS.BOCA[1].Codigo_Barras"',
      attributeId: AttributeIds.Value
 };
const dataValue = await session.read(nodeToRead, maxAge);
console.log(dataValue.toString());

{ /* DataValue */
value: Variant(Scalar, value: )
statusCode: BadNodeIdUnknown (0x80340000)
serverTimestamp: 2021-03-22T10:34:23.115Z $ 007.700.000
sourceTimestamp: null
}

AndreasHeine, 2021-03-24:
"BadNodeIdUnknown" is always a sign that the server you are connected to does not have this nodeid...

jordicolom, 2021-03-24:
Hello,
Thanks for reply.

I connect to a computer that is a copy of the original to carry out the tests, the only difference with the original is that it does not have the physical connection with Siemens S7 but it does have all the software installed and the OPC server. I have installed a UA Expert and I put a capture of the image where the nodeID appears 

[image]

AndreasHeine, 2021-03-24:
uaexpert shows the same result for attribute value!

jordicolom, 2021-03-24:
so is the return of the value really taking place?

I have tried to change the value with the following and it gives me an error too 

nodeID ='ns=7;s=S7300ET200M station_1.PLC_1.REGISTROS_BOCAS.BOCA[1].Codigo_Barras';
var dataToWrite = {
       dataType: "Int32",
       value: 25
 }
 let salida = await session.writeSingleNode(nodeId, dataToWrite)
console.log(salida);

ConstantStatusCode {
  _value: 2151350272,
  _description: 'The access level does not allow writing to the Node.',
  _name: 'BadNotWritable' }
{ /* DataValue */
   value: Variant(Scalar<Null>, value: <null>)
   statusCode:      BadAttributeIdInvalid (0x80350000)
   serverTimestamp: 2021-03-24T14:17:59.493Z $ 984.000.000
   sourceTimestamp: null
}

jordicolom, 2021-03-24:
If i write i have error.
Why if i read the nodeID with browseResult , this node is not posible write?
In the image is listed with browseResult and later if i write, return the error this node not exist in the server address?

[image]

AndreasHeine, 2021-03-24:
uaexpert returns the nodeidunknown for the value attribute (not for the node itself which is pretty weird) i would bet if you use attribute datatype it will retun a propper result

AndreasHeine, 2021-03-24:
there must be something wrong configured on siemens side...

jordicolom, 2021-03-24:
the same error with this:

var nodesToWrite = [{
            nodeId: nodo,
            attributeId: AttributeIds.Value,
            value: { 
                value: { 
                    dataType: DataType.Int,
                     value: 34
                }
          }
        }];
        session.write(nodesToWrite, function (err, statusCodes) {
            if (!err) {
                console.log(statusCodes);
            }
            console.log(err);
         });

jordicolom, 2021-03-24:
If i don't have connected the Siemens S7 to the computer is posible read and write varibles to the opc server?

AndreasHeine, 2021-03-24:
siemens... dont know!

jordicolom, 2021-03-24:
Ok thanks, I don't know if the problem is in the nodeID or because this computer is not connected to the siemens

AndreasHeine, 2021-03-24:
uaexpert is a good testing tool if it does not work with uaexpert its 98% chance its on server side

jordicolom, 2021-04-17:
Hello,
The problem is already solved. The problem was that the computer that had the opc server installed did not have the Siemens PLC physically connected. Once the test was carried out in the factory with the computer that had connected, the reading was correct, REGISTRO_MOCAS is the name of the node. The reading can be done by:

var node = 'ns = 7; s = S7300ET200M station_1.PLC_1.AUX_REGISTROS_MOCAS.BOCA [1] .Code_Barras';
const nodeToRead = {
             nodeId: node,
             attributeId: AttributeIds.Value
         };
         // test read
         const dataValue = await session.read (nodeToRead);
and writing by:

  var nodeWrite = "ns = 7; s = S7300ET200M station_1.PLC_1.PC.BOCA_DESTINO";

  var buffer = Buffer.from ("2");
         var nodesToWrite = {
             dataType: DataType.Byte,
             value: buffer
         }

output = await session.writeSingleNode (nodeWrite, nodesToWrite);

thanks

erossignon, 2021-04-24:
Let's close then,

closed 2021-04-24 by erossignon