node-opcua/node-opcua #971: Race condition on asynchronous creation of default certificate

prmdhost, 2021-03-25:
**Current behavior**

When the OPCUAClient is created and a default certificate does not exist, the client creates a new certificate in the home folder.

**Describe the bug**

However, when multiple clients are created at the same time, a race condition happens, where each client tries to create the certificate at the same time, thus leading to unexpected behavior.

```
Create client 0
Create client 1
Create client 2
Create client 3
Create client 4
Client 0 connected
Client 3 cannot connect to endpoint
[Error: ENOENT: no such file or directory, unlink '~\node-opcua-default-nodejs\Config\PKI\own\certs\client_certificate.pem.csr'] {
  errno: -4058,
  code: 'ENOENT',
  syscall: 'unlink',
  path: '~\\node-opcua-default-nodejs\\Config\\PKI\\own\\certs\\client_certificate.pem.csr'
}
Client 4 cannot connect to endpoint
[Error: ENOENT: no such file or directory, unlink '~\node-opcua-default-nodejs\Config\PKI\own\certs\client_certificate.pem.csr'] {
  errno: -4058,
  code: 'ENOENT',
  syscall: 'unlink',
  path: '~\\node-opcua-default-nodejs\\Config\\PKI\\own\\certs\\client_certificate.pem.csr'
}
Client 1 cannot connect to endpoint
[Error: ENOENT: no such file or directory, unlink '~\node-opcua-default-nodejs\Config\PKI\own\certs\client_certificate.pem.csr'] {
  errno: -4058,
  code: 'ENOENT',
  syscall: 'unlink',
  path: '~\\node-opcua-default-nodejs\\Config\\PKI\\own\\certs\\client_certificate.pem.csr'
}
Client 2 cannot connect to endpoint
[Error: ENOENT: no such file or directory, unlink '~\node-opcua-default-nodejs\Config\PKI\own\certs\client_certificate.pem.csr'] {
  errno: -4058,
  code: 'ENOENT',
  syscall: 'unlink',
  path: '~\\node-opcua-default-nodejs\\Config\\PKI\\own\\certs\\client_certificate.pem.csr'
}
```

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. Folder `~/node-opcua-default-nodejs` does not exist
2. Create multiple clients at the same time
3. Most of them will fail due to the race condition

**Expected behavior**

Creation of default certificate should be synchronized.

**Screenshots**

**Gist**

 
gist: https://gist.github.com/prmdhost/f4b83f1cfc5d4b6d099931657e42c45c

**Context**

- (X) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (X) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: Windows desktop and Docker containers

    - (X) Windows : version : Windows 10 Pro
    - (X) Linux : version : Debian 10
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: Prosys Simulation Server
    - Version: 5.0.4-284

- **node-opcua version:** : 2.36.0
  
  

- **Node:**
  `node --version` =v15.11.0

erossignon, 2021-03-29:
If the clients are all hosted inside the same executable:
 - make sure that the Certificated manager and client certificate get created and initialized before the clients start.
 - Share this certificate manager by providing it to the client constructor, so the default cm is not used.

If the clients are running on separate processes, I would recommend using a different PKI for each client.

prmdhost, 2021-03-30:
Ok,  but other than that, it might still be possible to run into the issue.

Maybe we could synchronize the call to the certificate manager by using a lock-file instead of relying on the certificateFile to exist.

```js
protected async createDefaultCertificate() {
        if (!fs.existsSync(this.certificateFile)) {
            ...
            await this.clientCertificateManager.createSelfSignedCertificate({...});
            ...
        }
        if (!fs.existsSync(this.certificateFile)) {
            throw new Error(" cannot locate certificate file " + this.certificateFile);
        }
    }
```

erossignon, 2021-04-23:
@prmdhost please test with node-opcua@2.40.0 and confirm that hte prob is now fixed.

prmdhost, 2021-04-26:
Thanks @erossignon, works perfectly

closed 2021-04-26 by prmdhost