node-opcua/node-opcua #972: Weird output Error on Install Win10 node-opcua "latest"

AndreasHeine, 2021-03-26:
> Microsoft Windows [Version 10.0.19042.867]
> (c) 2020 Microsoft Corporation. Alle Rechte vorbehalten.
> 
> C:\WINDOWS\system32>cd C:\Users\andre\Desktop\node-opcua-server-demo
> 
> C:\Users\andre\Desktop\node-opcua-server-demo>npm install node-opcua
> 
> > node-opcua-pki@2.8.0 postinstall C:\Users\andre\Desktop\node-opcua-server-demo\node_modules\node-opcua-pki
> > opencollective-postinstall
> 
> Der Befehl "opencollective-postinstall" ist entweder falsch geschrieben oder
> konnte nicht gefunden werden.
> npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@~2.3.1 (node_modules\chokidar\node_modules\fsevents):
> npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.3.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})
> npm WARN opcua-server-demo@1.0.0 No description
> npm WARN opcua-server-demo@1.0.0 No repository field.
> 
> npm ERR! code ELIFECYCLE
> npm ERR! errno 1
> npm ERR! node-opcua-pki@2.8.0 postinstall: `opencollective-postinstall`
> npm ERR! Exit status 1
> npm ERR!
> npm ERR! Failed at the node-opcua-pki@2.8.0 postinstall script.
> npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
> 
> npm ERR! A complete log of this run can be found in:
> npm ERR!     C:\Users\andre\AppData\Roaming\npm-cache\_logs\2021-03-26T17_15_26_629Z-debug.log

as reference on "2.21.0" no Errors on instalation!
does opencollective-postinstall do somthing important? ^^

erossignon, 2021-03-27:
I removed this package a while ago , but didn't reissue node-opcua-pki yet.

erossignon, 2021-03-27:
opencollective-postinstall  display a message to encourage user to sponsors us so we can keep developing node-opcua.

Is this consistently reproducible? On my side I have : 

````console
C:\temp\demo>npm install node-opcua

> node-opcua-pki@2.8.0 postinstall C:\temp\demo\node_modules\node-opcua-pki
> opencollective-postinstall

Thank you for using node-opcua-pki!
If you rely on this package, please consider supporting our open collective:
> https://opencollective.com/node-opcua/donate

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@~2.3.1 (node_modules\chokidar\node_modules\fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.3.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"win32","arch":"x64"})
npm WARN demo@1.0.0 No description
npm WARN demo@1.0.0 No repository field.

+ node-opcua@2.36.0
added 310 packages from 229 contributors and audited 312 packages in 13.175s

35 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
 ````

AndreasHeine, 2021-03-29:
it occurs not everytime.... sorry about that!

closed 2021-03-29 by AndreasHeine