node-opcua/node-opcua #975: 'pnpm install' fails on fresh workspace

5yn4ps3, 2021-03-30:
**Current behavior**
I want to build the client demo from the source code, like described in the readme.
I get the following errors at the 'pnpm install' step;

│ ../node-opcua-benchmarker/source/benchmarker.ts(8,24): error TS2307: Cannot find module 'node-opcua-assert' or its corresponding type declarations.

Note that I tried from `master` and `v2.35` with same results.

**Describe the bug**
Here is the full log;

> Microsoft Windows [version 10.0.19041.867]
(c) 2019 Microsoft Corporation. Tous droits réservés.
C:\Projects\node-opcua 2.35.0>**pnpm install**
Recreating C:\Projects\node-opcua 2.35.0\node_modules
Lockfile is up-to-date, resolution step is skipped
> node-opcua-mono-repo@2.9.1 preinstall C:\Projects\node-opcua 2.35.0
> node prevent_npm_install.js
Packages: +1321
+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++  
Packages are hard linked from the content-addressable store to the virtual store.
  Content-addressable store is at: C:\Users\SESA451761\.pnpm-store\v3
  Virtual store is at:             node_modules/.pnpm
Progress: resolved 1321, reused 1321, downloaded 0, added 1321, done
 WARN  Failed to find "/fsevents/2.3.2" in lockfile during hoisting. Next aliases will not be hoisted: fsevents
node_modules/.pnpm/core-js@2.6.12/node_modules/core-js: Running postinstall script, done in 419ms
node_modules/.pnpm/@fortawesome/fontawesome-common-types@0.2.34/node_modules/@fortawesome/fontawesome-common-types: Running postinstall script, done in 384ms
node_modules/.pnpm/protobufjs@6.10.2/node_modules/protobufjs: Running postinstall script, done in 420ms
node_modules/.pnpm/node-opcua-pki@2.8.0/node_modules/node-opcua-pki: Running postinstall script, done in 362ms
node_modules/.pnpm/@fortawesome/fontawesome-svg-core@1.2.34/node_modules/@fortawesome/fontawesome-svg-core: Running postinstall script, done in 381ms
node_modules/.pnpm/@fortawesome/free-solid-svg-icons@5.15.2/node_modules/@fortawesome/free-solid-svg-icons: Running postinstall script, done in 341ms
Cannot link binary 'eslint' of 'eslint-cli' to 'C:\Projects\node-opcua 2.35.0\node_modules\.bin': binary of 'eslint' is already linked
dependencies:
...
devDependencies:
...
> node-opcua-mono-repo@2.9.1 postinstall C:\Projects\node-opcua 2.35.0
> pnpm build
> node-opcua-mono-repo@2.9.1 build C:\Projects\node-opcua 2.35.0
> npm run generateWithPNPM && tsc -b packages
> node-opcua-mono-repo@2.9.1 generateWithPNPM C:\Projects\node-opcua 2.35.0
> pnpm recursive run generate
Scope: all 75 projects
packages/node-opcua-types generate$ tsc -b tsconfig.pre.json && ts-node source/generate && tsc -b
│ ../node-opcua-benchmarker/source/benchmarker.ts(8,24): **error TS2307: Cannot find module 'node-opcua-assert' or its corresponding type declarations.**
│ ../node-opcua-utils/source/watchdog.ts(5,24): **error TS2307: Cannot find module 'node-opcua-assert' or its corresponding type declarations.**
│ ../node-opcua-utils/source/set_deprecated.ts(6,24): **error TS2307: Cannot find module 'node-opcua-assert' or its corresponding type declarations.**
│ ../node-opcua-utils/source/index.ts(5,24): **error TS2307: Cannot find module 'node-opcua-assert' or its corresponding type declarations.**
└─ Failed in 3.3s
C:\Projects\node-opcua 2.35.0\packages\node-opcua-types:
 ERROR  node-opcua-types@2.35.0 generate: `tsc -b tsconfig.pre.json && ts-node source/generate && tsc -b`
Exit status 1
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-opcua-mono-repo@2.9.1 generateWithPNPM: `pnpm recursive run generate`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-opcua-mono-repo@2.9.1 generateWithPNPM script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\SESA451761\AppData\Roaming\npm-cache\_logs\2021-03-30T08_02_15_058Z-debug.log
 ERROR  Command failed with exit code 1.
 ERROR  Command failed with exit code 1.
C:\Projects\node-opcua 2.35.0>    

**Step by step instruction to reproduce**

1. `git clone https://github.com/node-opcua/node-opcua.git`
2. `cd node-opcua`
3. `npm install -g pnpm`
4. `pnpm install`

**Expected behavior**

Step 4. (`pnpm install`) should works on a fresh workspace.

**Context**

- (X) my request is related to node-opcua acting as an OPCUA CLIENT
- (X) my request is related to node-opcua acting as an OPCUA SERVER

- (X) I have installed node-opcua from source ( using git clone)
- ( ) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: PC
- OS version: Windows 10 x64

- **Node:**
  `node --version` = v14.9.0
  `npm --version` = v6.14.8
  `pnpm --version` = v5.18.9

5yn4ps3, 2021-03-30:
`lerna link` before the `pnpm build` seems to solve the issue.

erossignon, 2021-03-30:
run  `pnpm i -r`

closed 2021-03-30 by erossignon