node-opcua/node-opcua #976: method call example brocken

harrywg, 2021-04-01:
**I'm submitting a ...**  (check one with "x")

 - ( x) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request

 - ( ) support request 

**Current behavior**

server_with_method.js example crashes when method is called from UA-Expert 
**Expected behavior**

**Minimal reproduction of the problem with instructions**

  1. either clone repo or do npm install
  2. run  server_with_method.js
  3. server starts up fine and method is shwon in UA-Expert
  4. call method from UA-Expert, server throws exceptions
  ...

gist:

  
**What is the motivation / use case for changing the behavior?**

**Please tell us about your environment:**

  * ( ) my request is related to node-opcua acting as a  OPCUA CLIENT 
  * (x)  my request is related to node-opcua acting as a OPCUA SERVER 

  * (x) I have installed node-opcua from source ( using git clone)
  * (x ) I have installed node-opcua as a package ( using npm install )
  * ( ) I am using an application that uses node-opcua 
       - ( ) node-red
       - ( ) other : please specify 
        
  * Device: _____
  * OS version: _____
    
    - ( ) Windows : version : _________
    - (x) Linux   : version : docker node:14-alpine, node:12-alpine, node:14-slim, node:12-slim
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name:_________
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:**  :

* **Node:** 
   `node --version` = verified with 12 and 14

* **Typscript:** 
   `tsc --version` =
     

const { 
    OPCUAServer,
    DataType,
    Variant,VariantArrayType,
    StatusCodes,
    makeAccessLevelFlag 
} = require("node-opcua");

[... 96 lines cut ...]
err TypeError: Cannot read property 'getExtensionObjectConstructorFromBinaryEncoding' of undefined
    at resolveDynamicExtensionObjectV (/app/node-opcua/packages/node-opcua-client-dynamic-extension-object/dist/resolve_dynamic_extension_object.js:10:45)
    at /app/node-opcua/packages/node-opcua-client-dynamic-extension-object/dist/resolve_dynamic_extension_object.js:52:29
    at Array.map (<anonymous>)
    at Object.resolveDynamicExtensionObject (/app/node-opcua/packages/node-opcua-client-dynamic-extension-object/dist/resolve_dynamic_extension_object.js:51:43)
    at performWrite (/app/node-opcua/packages/node-opcua-server/dist/server_engine.js:986:58)
    at /app/node-opcua/packages/node-opcua-server/node_modules/.pnpm/async@3.2.0/node_modules/async/dist/async.js:243:13
    at eachOfArrayLike (/app/node-opcua/packages/node-opcua-server/node_modules/.pnpm/async@3.2.0/node_modules/async/dist/async.js:500:13)
    at eachOf (/app/node-opcua/packages/node-opcua-server/node_modules/.pnpm/async@3.2.0/node_modules/async/dist/async.js:551:16)
    at awaitable (/app/node-opcua/packages/node-opcua-server/node_modules/.pnpm/async@3.2.0/node_modules/async/dist/async.js:208:32)
    at _asyncMap (/app/node-opcua/packages/node-opcua-server/node_modules/.pnpm/async@3.2.0/node_modules/async/dist/async.js:241:16)
(node:27) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'toString' of undefined
    at resolveDynamicExtensionObjectV (/app/node-opcua/packages/node-opcua-client-dynamic-extension-object/dist/resolve_dynamic_extension_object.js:36:57)
    at /app/node-opcua/packages/node-opcua-client-dynamic-extension-object/dist/resolve_dynamic_extension_object.js:52:29
    at Array.map (<anonymous>)
    at Object.resolveDynamicExtensionObject (/app/node-opcua/packages/node-opcua-client-dynamic-extension-object/dist/resolve_dynamic_extension_object.js:51:43)
    at performWrite (/app/node-opcua/packages/node-opcua-server/dist/server_engine.js:986:58)
    at /app/node-opcua/packages/node-opcua-server/node_modules/.pnpm/async@3.2.0/node_modules/async/dist/async.js:243:13
    at eachOfArrayLike (/app/node-opcua/packages/node-opcua-server/node_modules/.pnpm/async@3.2.0/node_modules/async/dist/async.js:500:13)
    at eachOf (/app/node-opcua/packages/node-opcua-server/node_modules/.pnpm/async@3.2.0/node_modules/async/dist/async.js:551:16)
    at awaitable (/app/node-opcua/packages/node-opcua-server/node_modules/.pnpm/async@3.2.0/node_modules/async/dist/async.js:208:32)
    at _asyncMap (/app/node-opcua/packages/node-opcua-server/node_modules/.pnpm/async@3.2.0/node_modules/async/dist/async.js:241:16)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:27) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:27) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

erossignon, 2021-04-05:
the code you provided works on my side :
````
node issue_976.js  
Debugger attached.
Server is now listening ... ( press CTRL+C to stop)
 the primary server endpoint url is  opc.tcp://DESKTOP-F0VVBQ0:4334
Hello World ! I will bark  1  times
the requested volume is  1 
Hello World ! I will bark  1  times
the requested volume is  1 
Hello World ! I will bark  1  times
the requested volume is  1 
Hello World ! I will bark  1  times
the requested volume is  1 
````

harrywg, 2021-04-05:
Etienne,
thx for your answer.
I pulled the latest version from Github.
Is there a need for a specific version of node?
Any good entry point to debugging?
Thx in advance.

err TypeError: Cannot read property 'namespace' of undefined
    at ExtraDataTypeManager.getExtensionObjectConstructorFromBinaryEncoding (/app/node_modules/node-opcua-client-dynamic-extension-object/dist/extra_data_type_manager.js:50:90)
    at resolveDynamicExtensionObjectV (/app/node_modules/node-opcua-client-dynamic-extension-object/dist/resolve_dynamic_extension_object.js:10:45)
    at /app/node_modules/node-opcua-client-dynamic-extension-object/dist/resolve_dynamic_extension_object.js:52:29
    at Array.map (<anonymous>)
    at Object.resolveDynamicExtensionObject (/app/node_modules/node-opcua-client-dynamic-extension-object/dist/resolve_dynamic_extension_object.js:51:43)
    at performWrite (/app/node_modules/node-opcua-server/dist/server_engine.js:986:58)
    at /app/node_modules/async/dist/async.js:243:13
    at eachOfArrayLike (/app/node_modules/async/dist/async.js:500:13)
    at eachOf (/app/node_modules/async/dist/async.js:551:16)
    at awaitable (/app/node_modules/async/dist/async.js:208:32)
err TypeError: Cannot read property 'getExtensionObjectConstructorFromBinaryEncoding' of undefined
    at resolveDynamicExtensionObjectV (/app/node_modules/node-opcua-client-dynamic-extension-object/dist/resolve_dynamic_extension_object.js:10:45)
    at /app/node_modules/node-opcua-client-dynamic-extension-object/dist/resolve_dynamic_extension_object.js:52:29
    at Array.map (<anonymous>)
    at Object.resolveDynamicExtensionObject (/app/node_modules/node-opcua-client-dynamic-extension-object/dist/resolve_dynamic_extension_object.js:51:43)
    at performWrite (/app/node_modules/node-opcua-server/dist/server_engine.js:986:58)
    at /app/node_modules/async/dist/async.js:243:13
    at eachOfArrayLike (/app/node_modules/async/dist/async.js:500:13)
    at eachOf (/app/node_modules/async/dist/async.js:551:16)
    at awaitable (/app/node_modules/async/dist/async.js:208:32)
    at _asyncMap (/app/node_modules/async/dist/async.js:241:16)
(node:1) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'toString' of undefined
    at resolveDynamicExtensionObjectV (/app/node_modules/node-opcua-client-dynamic-extension-object/dist/resolve_dynamic_extension_object.js:36:57)
    at /app/node_modules/node-opcua-client-dynamic-extension-object/dist/resolve_dynamic_extension_object.js:52:29
    at Array.map (<anonymous>)
    at Object.resolveDynamicExtensionObject (/app/node_modules/node-opcua-client-dynamic-extension-object/dist/resolve_dynamic_extension_object.js:51:43)
    at performWrite (/app/node_modules/node-opcua-server/dist/server_engine.js:986:58)
    at /app/node_modules/async/dist/async.js:243:13
    at eachOfArrayLike (/app/node_modules/async/dist/async.js:500:13)
    at eachOf (/app/node_modules/async/dist/async.js:551:16)
    at awaitable (/app/node_modules/async/dist/async.js:208:32)
    at _asyncMap (/app/node_modules/async/dist/async.js:241:16)
    at emitUnhandledRejectionWarning (internal/process/promises.js:168:15)
    at processPromiseRejections (internal/process/promises.js:247:11)
    at processTicksAndRejections (internal/process/task_queues.js:94:32)
(node:1) TypeError: Cannot read property 'toString' of undefined
    at resolveDynamicExtensionObjectV (/app/node_modules/node-opcua-client-dynamic-extension-object/dist/resolve_dynamic_extension_object.js:36:57)
    at /app/node_modules/node-opcua-client-dynamic-extension-object/dist/resolve_dynamic_extension_object.js:52:29
    at Array.map (<anonymous>)
    at Object.resolveDynamicExtensionObject (/app/node_modules/node-opcua-client-dynamic-extension-object/dist/resolve_dynamic_extension_object.js:51:43)
    at performWrite (/app/node_modules/node-opcua-server/dist/server_engine.js:986:58)
    at /app/node_modules/async/dist/async.js:243:13
    at eachOfArrayLike (/app/node_modules/async/dist/async.js:500:13)
    at eachOf (/app/node_modules/async/dist/async.js:551:16)
    at awaitable (/app/node_modules/async/dist/async.js:208:32)
    at _asyncMap (/app/node_modules/async/dist/async.js:241:16)
(node:1) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a

    at emitDeprecationWarning (internal/process/promises.js:180:11)
    at processPromiseRejections (internal/process/promises.js:249:13)
    at processTicksAndRejections (internal/process/task_queues.js:94:32)

Antworten an: node-opcua/node-opcua ***@***.***>
Datum: Montag, 5. April 2021 um 18:30
An: node-opcua/node-opcua ***@***.***>
Cc: Harald Weigold ***@***.***>, Author ***@***.***>

erossignon, 2021-04-05:
I have tried from a fresh clone:

```bash
$ git clone https://github.com/node-opcua/node-opcua
$ npm install -g pnpm
$ cd node-opcua
$ pnpm recursive install
$ node documentation/server_with_method.js
```` 

and get the expected result.

What's different on your end?

harrywg, 2021-04-06:
Hi Etienne,
I got it working with my own client written with node-opcua.
That's fine for me now.

Thx a lot for your help and your amazing work on node-opcua.

Harald

Datum: Montag, 5. April 2021 um 23:36
Cc: Harald Weigold ***@***.***>, Assign ***@***.***>

harrywg, 2021-04-06:
works after changing the client

closed 2021-04-06 by harrywg