node-opcua/node-opcua #978: Shutting down the Node OPC UA server and starting it after leads to exception in opcua_server.js file line 677

Graos, 2021-04-06:
Shutting down the Node OPC UA server and starting it after leads to exception in opcua_server.js file line 677

Start the Nodeopcua server.
Client establishes connection to the server.
shutdown the nodeopcua server by calling shutdown function of the server class.
Start the server by calling tart function of the server class.
nodeopcua server gets into exception

Trace Info:

node:8640) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'stop' of undefined
    at OPCUAServer.shutdown (D:\Project\nodeopcuaServer\node_modules\node-opcua-server\dist\opcua_server.js:677:36)
    at OPCUAServer.shutdown (D:\Project\nodeopcuaServer\node_modules\thenify\index.js:65:46)
    at D:\Project\nodeopcuaServer\node_modules\node-opcua-server\dist\opcua_server.js:641:26
    at D:\Project\nodeopcuaServer\node_modules\node-opcua-server\dist\base_server.js:173:29
(Use `node --trace-warnings ...` to show where the warning was created)
(node:8640) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:8640) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

Code Gist
        if (TestOpcUaServer.self.testControlNodes.EnableIntermittentConnectionSimulation.connectionState == "Shutdown") {
            TestOpcUaServer.self.testControlNodes.EnableIntermittentConnectionSimulation.connectionState = "StartingServer";
            TestOpcUaServer.self.server.start(() => {
                console.log(`${TestOpcUaServer.self.getTimeStampStringWithSeverity("Info")} --> Server started with state : ${TestOpcUaServer.self.server.engine.serverStatus.state}`);
                TestOpcUaServer.self.testControlNodes.EnableIntermittentConnectionSimulation.connectionState = "Running";
            });
        }
        else {
            if (TestOpcUaServer.self.testControlNodes.EnableIntermittentConnectionSimulation.connectionState == "Running") {
                TestOpcUaServer.self.testControlNodes.EnableIntermittentConnectionSimulation.connectionState == "ShutdownInitiated"
                TestOpcUaServer.self.server.shutdown(function (err) {
                    console.log(TestOpcUaServer.self.getTimeStampStringWithSeverity("Info") + " --> " + "Server shutdown ");
                    TestOpcUaServer.self.testControlNodes.EnableIntermittentConnectionSimulation.connectionState = "Shutdown"
                });
            }
        }

- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua as a package ( using npm install )

- Device: _Windows_____
- OS version: _10_____

    - ( ) Windows : version : ______
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: __UaExpert____
    - Version: ___1.5.1___
    - Manufacturer/Software vendor: _UA_____
    - link to the manufacturer or product site: https://

- **node-opcua version:** : 2.34
  
  

- **Node:**
  `node --version` = v14.6

erossignon, 2021-04-06:
Once shutdown, a server cannot be restarted. you need to recreate a new server and start it. That should be a simple work around.

Graos, 2021-04-07:
Ok Thanks.

closed 2021-04-07 by Graos