node-opcua/node-opcua #980: Module '"node-opcua"' has no exported member 'ReadValueIdLike

oneproxima, 2021-04-14:
**Current behavior**

I receive the following error while following the simple client tutorial: Module '"node-opcua"' has no exported member 'ReadValueIdLike'.

**Describe the bug**

It's a compiler error as mentioned above.

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. Followed the tutorial and went to run the script below.

**Expected behavior**
Expected at least the program to run and not a compile error.

**Screenshots**

sample_client_ts.ts:9:5 - error TS2305: Module '"node-opcua"' has no exported member 'ReadValueIdLike'.

9     ReadValueIdLike,
      ~~~~~~~~~~~~~~~

    at createTSError (C:\Users\forna\AppData\Roaming\npm\node_modules\ts-node\src\index.ts:513:12)
    at reportTSError (C:\Users\forna\AppData\Roaming\npm\node_modules\ts-node\src\index.ts:517:19)
    at getOutput (C:\Users\forna\AppData\Roaming\npm\node_modules\ts-node\src\index.ts:752:36)
    at Object.compile (C:\Users\forna\AppData\Roaming\npm\node_modules\ts-node\src\index.ts:968:32)
    at Module.m._compile (C:\Users\forna\AppData\Roaming\npm\node_modules\ts-node\src\index.ts:1056:42)
    at Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
    at Object.require.extensions.<computed> [as .ts] (C:\Users\forna\AppData\Roaming\npm\node_modules\ts-node\src\index.ts:1059:12)
    at Module.load (internal/modules/cjs/loader.js:928:32)
    at Function.Module._load (internal/modules/cjs/loader.js:769:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:72:12)

**Gist**

 
gist:

`import {
   OPCUAClient,
   MessageSecurityMode, SecurityPolicy,
   AttributeIds,
   makeBrowsePath,
   ClientSubscription,
   TimestampsToReturn,
   MonitoringParametersOptions,
   ReadValueIdLike,
   ClientMonitoredItem,
   DataValue
} from "node-opcua";

const connectionStrategy = {
    initialDelay: 1000,
    maxRetry: 1
}
const options = {
    applicationName: "MyClient",
    connectionStrategy: connectionStrategy,
    securityMode: MessageSecurityMode.None,
    securityPolicy: SecurityPolicy.None,
    endpoint_must_exist: false,
};
const client = OPCUAClient.create(options);
// const endpointUrl = "opc.tcp://opcuademo.sterfive.com:26543";
const endpointUrl = "opc.tcp://localhost:4840";
async function main() {
  try {
    // step 1 : connect to
        await client.connect(endpointUrl);
        console.log("connected !");

    // step 2 : createSession
        const session = await client.createSession();
        console.log("session created !");

    // step 3 : browse
        const browseResult = await session.browse("RootFolder");

        console.log("references of RootFolder :");
        for(const reference of browseResult.references) {
            console.log( "   -> ", reference.browseName.toString());
        }

    // step 4 : read a variable with readVariableValue
        const dataValue2 = await session.readVariableValue("ns=3;s=Scalar_Simulation_Double");
        console.log(" value = " , dataValue2.toString());

    // step 4' : read a variable with read
        const maxAge = 0;
        const nodeToRead = {
          nodeId: "ns=3;s=Scalar_Simulation_String",
          attributeId: AttributeIds.Value
        };
        const dataValue =  await session.read(nodeToRead, maxAge);
        console.log(" value " , dataValue.toString());

    // step 5: install a subscription and install a monitored item for 10 seconds
    const subscription = ClientSubscription.create(session, {
        requestedPublishingInterval: 1000,
        requestedLifetimeCount:      100,
        requestedMaxKeepAliveCount:   10,
        maxNotificationsPerPublish:  100,
        publishingEnabled: true,
        priority: 10
    });

    subscription.on("started", function() {
        console.log("subscription started for 2 seconds - subscriptionId=", subscription.subscriptionId);
    }).on("keepalive", function() {
        console.log("keepalive");
    }).on("terminated", function() {
       console.log("terminated");
    });

    // install monitored item

    const itemToMonitor: ReadValueIdLike = {
        nodeId: "ns=3;s=Counter1a",
        attributeId: AttributeIds.Value
    };
    const parameters: MonitoringParametersOptions = {
        samplingInterval: 100,
        discardOldest: true,
        queueSize: 10
    };

    const monitoredItem  = ClientMonitoredItem.create(
        subscription,
        itemToMonitor,
        parameters,
        TimestampsToReturn.Both
    );

    monitoredItem.on("changed", (dataValue: DataValue) => {
       console.log(" value has changed : ", dataValue.value.toString());
    });

    async function timeout(ms: number) {
        return new Promise(resolve => setTimeout(resolve, ms));
    }
    await timeout(10000);

    console.log("now terminating subscription");
    await subscription.terminate();

    // step 6: finding the nodeId of a node by Browse name
        const browsePath = makeBrowsePath("RootFolder", "/Objects/Server.ServerStatus.BuildInfo.ProductName");

        const result = await session.translateBrowsePath(browsePath);
        const productNameNodeId = result.targets[0].targetId;
        console.log(" Product Name nodeId = ", productNameNodeId.toString());

    // close session
        await session.close();

    // disconnecting
        await client.disconnect();
        console.log("done !");
  } catch(err) {
    console.log("An error has occured : ",err);
  }
}
main();`

**Context**

- (x) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: Dell XPS 15
- OS version: Windows 10

    - (x) Windows : version : 20H2
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name:  RSLinx Classic Gateway
    - Version: 4.21
    - Manufacturer/Software vendor: Rockwell Automation
    - link to the manufacturer or product site: https://

- **node-opcua version:** :
  
  

- **Node:**
  `node --version` = 14.15.5

erossignon, 2021-04-14:
This has been fixed but not published to main branch yet.

erossignon, 2021-04-14:
duplicates #965

closed 2021-04-14 by erossignon