node-opcua/node-opcua #985: Crash upon automatic reconnection with new server certificate

paolodenti, 2021-04-23:
opcuaserver crashes with the message below

```
21:08:38.628Z :base_server                   :464    messages  EXCEPTION CAUGHT WHILE PROCESSING REQUEST !!! ActivateSessionRequest
 ...                                                Attempt to access memory outside buffer bounds
 ...                                                RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds
 ...                                                    at boundsError (internal/buffer.js:81:11)
 ...                                                    at Buffer.readUInt32LE (internal/buffer.js:217:5)
 ...                                                    at OPCUAServer.userNameIdentityTokenAuthenticateUser (<my folder>>/node_modules/node-opcua-server/dist/opcua_server.js:880:33)
 ...                                                    at OPCUAServer.isUserAuthorized (<my folder>>/node_modules/node-opcua-server/dist/opcua_server.js:930:25)
 ...                                                    at <my folder>>/node_modules/node-opcua-server/dist/opcua_server.js:1306:20
 ...                                                    at OPCUAServer.isValidUserNameIdentityToken (<my folder>>/node_modules/node-opcua-server/dist/opcua_server.js:795:16)
 ...                                                    at OPCUAServer.isValidUserIdentityToken (<my folder>>/node_modules/node-opcua-server/dist/opcua_server.js:907:25)
 ...                                                    at OPCUAServer._on_ActivateSessionRequest (<my folder>>/node_modules/node-opcua-server/dist/opcua_server.js:1295:16)
 ...                                                    at OPCUAServer.on_request (<my folder>>/node_modules/node-opcua-server/dist/base_server.js:288:25)
 ...                                                    at OPCUAServerEndPoint.<anonymous> (<my folder>>/node_modules/node-opcua-server/dist/opcua_server.js:503:26)
21:08:43.873Z :base_server                   :301   [NODE-OPCUA-W08] EXCEPTION CAUGHT WHILE PROCESSING REQUEST !! ActivateSessionRequest
21:08:43.873Z :base_server                   :302   { /*ActivateSessionRequest*/
 ...                                                 requestHeader                 /* RequestHeader       */: {
 ...                                                   authenticationToken         /* NodeId              */: ns=0;b=94c1addc65b7c5a5ce5ec8e5d57aa15e
 ...                                                   timestamp                   /* DateTime            */: 2021-04-23T21:08:41.555Z
 ...                                                   requestHandle               /* UInt32              */: 1000040               0xf4268
 ...                                                   returnDiagnostics           /* UInt32              */: 0               0x0
 ...                                                   auditEntryId                /* UAString            */: null
 ...                                                   timeoutHint                 /* UInt32              */: 10000               0x2710
 ...                                                   additionalHeader            /* ExtensionObject     */: null
 ...                                                 }
 ...                                                 clientSignature               /* SignatureData       */: {
 ...                                                   algorithm                   /* UAString            */: null
 ...                                                   signature                   /* ByteString          */: null
 ...                                                 }
 ...                                                 clientSoftwareCertificates    /* SignedSoftwareCer[] */: [ /* empty */ ]
 ...                                                 localeIds                     /* UAString         [] */: [ /* length =1*/
 ...                                                   en-US
 ...                                                 ]
 ...                                                 userIdentityToken             /* ExtensionObject     */: { /*UserNameIdentityToken*/
 ...                                                 policyId                      /* UAString            */: username_basic256Sha256
 ...                                                 userName                      /* UAString            */: testusername
 ...                                                 password                      /* ByteString          */
 ...                                                BUFFER
 ...                                                 encryptionAlgorithm           /* UAString            */: http://www.w3.org/2001/04/xmlenc#rsa-oaep
 ...                                                };
 ...                                                 userTokenSignature            /* SignatureData       */: {
 ...                                                 .... TRUNCATED .....
 display_trace_from_this_project_only = 
Attempt to access memory outside buffer bounds

```

## how to reproduce

1) start a server instance
2) connect with a client (e.g. UA Expert)
3) stop the server instance
4) restart the server instance on the same address using a docker image running the same server instance exposing the same port
5) the client fails the connection (expected behavior) with badsessionid, badinternalerror messages
6) the server instance, after some of the messages above, crashes

It looks the behavior has been there for some time, even with previous version the behavior is the same.

Reproducible with the last version (2.40.0)

The problem is not in the docker container. Even the other way around (start docker, stop, start local) crashes the server instance.

The same behavior can be reproduced with any client. I have the same issue both with UA Expert and with a client app using OPCUAClient.

I am available of course to run further analysis if addressed to the right direction.

paolodenti, 2021-04-23:
update:

upgrading node-opcua-client from 2.31 to 2.40, the client behavior changed and instead of retrying is disconnects, avoiding the server crash

```
21:36:55.986Z :client_base_impl              :351   the server certificate has changed,  we need to retrieve server certificate again:  ClientSecureChannelLayer => Socket is closed !
21:36:55.986Z :client_base_impl              :352   old server certificate  undefined
21:36:56.002Z :client_base_impl              :362   new server certificate  450a1a88afa0396ef26eefbafb287cbc8f0b7018

```

paolodenti, 2021-04-24:
it looks the issue is in the `userNameIdentityTokenAuthenticateUser` function in `opcua_server`.

When executing 
`const buff = cryptoFactory.asymmetricDecrypt(password, serverPrivateKey);
`with an invalid certificate, the returned value is `<Buffer 00>`
`and the following `buff.readUInt32LE(0)` fails

as a quick fix, just adding below
```
if (buff.length <= 1) {
    async.setImmediate(() => callback(null, false));
    return;
}

```
fixes the issue

erossignon, 2021-04-25:
@paolodenti , when using node-opcua inside a docker container , you have to make sure that the certifcates PKIs are store on a persitient volume or a volume share. this will prevent node-opcua to recreate new private keys and new certifices every time you start a container;

paolodenti, 2021-04-25:
@erossignon the issue is not regarding how the client is using the server. The client behavior was expected and desired. The issue is regarding the fact that any malicious client could crash any server just presenting wrong keys.

erossignon, 2021-04-25:
I get that. Nevertheless, I also fix an issue in the client reconnection as it could lead to presenting a badly encrypted password to the server... On stone , 2 birds.

erossignon, 2021-04-25:
this should now be fixed in node-opcua@2.41.0 , @paolodenti  please confirm.

paolodenti, 2021-04-25:
@erossignon tested and confirmed as fixed, thank you

closed 2021-05-08 by erossignon