node-opcua/node-opcua #990: "automaticallyAcceptUnknownCertificate" option is not respected

maximivanov, 2021-04-29:
**Current behavior**

When configuring a client with `clientCertificateManager` having the `automaticallyAcceptUnknownCertificate` option, it is not respected.

**Describe the bug**

Even if `automaticallyAcceptUnknownCertificate` is set to `true`, client connection to a server with self-signed certificate fails with 

> [NODE-OPCUA-W25] client's server certificate verification has failed  server Certificate verification failed with err Good (0x00000)

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

Reproduce with this code:

```js
import {
  OPCUAClient,
  MessageSecurityMode,
  SecurityPolicy,
} from 'node-opcua-client'
import { OPCUACertificateManager } from 'node-opcua-certificate-manager'

const endpointUrl = 'opc.tcp://opcuaserver.com:48010' // public demo server

;(async () => {
  const client = OPCUAClient.create({
    applicationName: 'MyClient',
    connectionStrategy: {
      initialDelay: 1000,
      maxRetry: 1,
    },
    clientCertificateManager: new OPCUACertificateManager({
      automaticallyAcceptUnknownCertificate: true,
    }),
    securityMode: MessageSecurityMode.SignAndEncrypt,
    securityPolicy: SecurityPolicy.Basic256Sha256,
    endpoint_must_exist: false,
    requestedSessionTimeout: 5 * 1000,
  })

  await client.connect(endpointUrl)

  const session = await client.createSession()

  const browseResult = await session.browse('RootFolder')

  console.log(`Children of RootFolder:`)
  for (const reference of browseResult.references) {
    console.log(
      ` → ${reference.browseName.toString()} // ${reference.nodeId.toString()}`,
    )
  }

  // close session
})()
```

If I comment out the `clientCertificateManager` option, the client connects and browses the root folder fine.

`clientCertificateManager.automaticallyAcceptUnknownCertificate = true` and `clientCertificateManager.automaticallyAcceptUnknownCertificate = false` behave identically - the client cannot connect and errors with message from above.

**Expected behavior**

Client trusts the server's self-signed certificate when automaticallyAcceptUnknownCertificate = true and errors when automaticallyAcceptUnknownCertificate = false.

**Context**

- (x) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: __OSX____
- OS version: __Big Sur 11.2.3____

    - ( ) Windows : version : ______
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: ___Public demo OPC server___
    - Version: ______
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: `opc.tcp://opcuaserver.com:48010`

- **node-opcua version:** :

node-opcua-certificate-manager: 2.41.1
node-opcua-client: 2.40.0

- **Node:**
  `node --version` = 14.16.1

erossignon, 2021-05-08:
please also provide package.json file, I suspect a version mismatch between two node-opcua modules.

erossignon, 2021-05-11:
I see that you have fix the version number in the package.json file and have rerun `npm install`

If you run  `npm list` , you will probably noticed that some node-opcua remains duplicated, this is an issue with npm.

running `npm dedupe`  should fix the issue.

Alternatively, remove the node_module folder entirely and rerun `npm install`