node-opcua/node-opcua #991: nodejs exits without further information whil parsing a custom nodeset.xml

rbeex, 2021-05-01:
**I'm submitting a ...**  (check one with "x")

 - ( ) bug report => search github for a similar issue or PR before submitting
 - ( ) feature request

 - ( x ) support request 

**Current behavior**

I implement a simple uaserver and try to load a custom nodeset (informationmodel) with >190 nodes modeled and checked with UAModeler. Until loading, the uaserver/nodejs exits without any further information, also no javascript exception is thrown who i can catch. 

The debug log shows:

 Setting debug for  factories_factories                                           to  true ALL
 Setting debug for  datatype_factory                                              to  true ALL
20:41:16.987Z :datatype_factory              :262   registering typeName  EUInformation ns=0;i=887
20:41:16.987Z :datatype_factory              :229    associateWithBinaryEncoding  EUInformation ns=0;i=889
20:41:16.987Z :datatype_factory              :262   registering typeName  Range ns=0;i=884
20:41:16.987Z :datatype_factory              :229    associateWithBinaryEncoding  Range ns=0;i=886
20:41:16.988Z :datatype_factory              :262   registering typeName  AxisInformation ns=0;i=12079
20:41:16.988Z :datatype_factory              :229    associateWithBinaryEncoding  AxisInformation ns=0;i=12089
20:41:16.991Z :datatype_factory              :262   registering typeName  FilterOperand ns=0;i=589
20:41:16.992Z :datatype_factory              :229    associateWithBinaryEncoding  FilterOperand ns=0;i=591
20:41:16.992Z :datatype_factory              :262   registering typeName  SimpleAttributeOperand ns=0;i=601
20:41:16.992Z :datatype_factory              :229    associateWithBinaryEncoding  SimpleAttributeOperand ns=0;i=603
20:41:16.993Z :datatype_factory              :262   registering typeName  RequestHeader ns=0;i=389
20:41:16.993Z :datatype_factory              :229    associateWithBinaryEncoding  RequestHeader ns=0;i=391
20:41:16.993Z :datatype_factory              :262   registering typeName  SignatureData ns=0;i=456
20:41:16.993Z :datatype_factory              :229    associateWithBinaryEncoding  SignatureData ns=0;i=458
20:41:16.994Z :datatype_factory              :262   registering typeName  SignedSoftwareCertificate ns=0;i=344
20:41:16.994Z :datatype_factory              :229    associateWithBinaryEncoding  SignedSoftwareCertificate ns=0;i=346
20:41:16.994Z :datatype_factory              :262   registering typeName  ActivateSessionRequest ns=0;i=465
20:41:16.994Z :datatype_factory              :229    associateWithBinaryEncoding  ActivateSessionRequest ns=0;i=467
20:41:16.995Z :datatype_factory              :262   registering typeName  ResponseHeader ns=0;i=392
20:41:16.995Z :datatype_factory              :229    associateWithBinaryEncoding  ResponseHeader ns=0;i=394
20:41:16.995Z :datatype_factory              :262   registering typeName  ActivateSessionResponse ns=0;i=468
20:41:16.995Z :datatype_factory              :229    associateWithBinaryEncoding  ActivateSessionResponse ns=0;i=470
20:41:16.996Z :datatype_factory              :262   registering typeName  AddNodesItem ns=0;i=376
20:41:16.996Z :datatype_factory              :229    associateWithBinaryEncoding  AddNodesItem ns=0;i=378
20:41:16.996Z :datatype_factory              :262   registering typeName  AddNodesRequest ns=0;i=486
20:41:16.996Z :datatype_factory              :229    associateWithBinaryEncoding  AddNodesRequest ns=0;i=488
20:41:16.997Z :datatype_factory              :262   registering typeName  AddNodesResult ns=0;i=483
20:41:16.997Z :datatype_factory              :229    associateWithBinaryEncoding  AddNodesResult ns=0;i=485
20:41:16.997Z :datatype_factory              :262   registering typeName  AddNodesResponse ns=0;i=489
20:41:16.997Z :datatype_factory              :229    associateWithBinaryEncoding  AddNodesResponse ns=0;i=491
20:41:16.998Z :datatype_factory              :262   registering typeName  AddReferencesItem ns=0;i=379
20:41:16.998Z :datatype_factory              :229    associateWithBinaryEncoding  AddReferencesItem ns=0;i=381
20:41:16.998Z :datatype_factory              :262   registering typeName  AddReferencesRequest ns=0;i=492
20:41:16.998Z :datatype_factory              :229    associateWithBinaryEncoding  AddReferencesRequest ns=0;i=494
20:41:16.999Z :datatype_factory              :262   registering typeName  AddReferencesResponse ns=0;i=495
20:41:16.999Z :datatype_factory              :229    associateWithBinaryEncoding  AddReferencesResponse ns=0;i=497
20:41:16.999Z :datatype_factory              :262   registering typeName  KeyValuePair ns=0;i=14533
20:41:16.999Z :datatype_factory              :229    associateWithBinaryEncoding  KeyValuePair ns=0;i=14846
20:41:17.000Z :datatype_factory              :262   registering typeName  AdditionalParametersType ns=0;i=16313
20:41:17.000Z :datatype_factory              :229    associateWithBinaryEncoding  AdditionalParametersType ns=0;i=17537
20:41:17.000Z :datatype_factory              :262   registering typeName  AggregateConfiguration ns=0;i=948
20:41:17.001Z :datatype_factory              :229    associateWithBinaryEncoding  AggregateConfiguration ns=0;i=950
20:41:17.001Z :datatype_factory              :262   registering typeName  MonitoringFilter ns=0;i=719
20:41:17.001Z :datatype_factory              :229    associateWithBinaryEncoding  MonitoringFilter ns=0;i=721
20:41:17.001Z :datatype_factory              :262   registering typeName  AggregateFilter ns=0;i=728
20:41:17.001Z :datatype_factory              :229    associateWithBinaryEncoding  AggregateFilter ns=0;i=730
20:41:17.002Z :datatype_factory              :262   registering typeName  MonitoringFilterResult ns=0;i=731
20:41:17.002Z :datatype_factory              :229    associateWithBinaryEncoding  MonitoringFilterResult ns=0;i=733
20:41:17.002Z :datatype_factory              :262   registering typeName  AggregateFilterResult ns=0;i=737
20:41:17.002Z :datatype_factory              :229    associateWithBinaryEncoding  AggregateFilterResult ns=0;i=739
20:41:17.002Z :datatype_factory              :262   registering typeName  AliasNameDataType ns=0;i=23468
20:41:17.003Z :datatype_factory              :229    associateWithBinaryEncoding  AliasNameDataType ns=0;i=23499
20:41:17.003Z :datatype_factory              :262   registering typeName  Annotation ns=0;i=891
20:41:17.003Z :datatype_factory              :229    associateWithBinaryEncoding  Annotation ns=0;i=893
20:41:17.003Z :datatype_factory              :262   registering typeName  UserIdentityToken ns=0;i=316
[... 672 lines cut ...]
        
  * Device: PC
  * OS version: Windows 10
    
    - ( x ) Windows : version : 10.0.19042Build 19042
    - ( ) Linux   : version : _________
    - ( ) MacOs   : version : _________
    - ( ) Raspbian: version : _________
    - ( ) Other   : specify :
   
   * Description of the other OPCUA system I am trying to connect to:
   
     - Name: none
     - Version:_________
     - Manufacturer/Software vendor:_________
     - link : https://
     
    
* **node-opcua version:**  : 2.41.1

* **Node:** 
   `node --version` = v14.16.1

* **Typscript:** 
   `tsc --version` = Version 1.0.3.0

erossignon, 2021-05-08:
Please contact support so that the specific nodeset could be investigated under confidentiality.

jmathieu-riis, 2021-08-16:
I am also having this same behavior when importing an xml file with some UAVariables that have German descriptions. One example of these descriptions is as follows: 
``` xml<Description Locale="de-DE">1=Werkstückträger in Modul-Position</Description>`
Adding that line or something similar to any of the UAVariable elements in a valid OPCUA xml file will cause node-opcua's xml parser to not close properly. It also does not print any error, it just exits the program in the middle of `server.initialize()`.

A temporary fix my team has found is to simply replace those characters with non-decorated characters/remove them entirely.

closed 2023-02-27 by erossignon