node-opcua/node-opcua #995: Can I used the node sdk in a react app?

danhellem, 2021-05-07:
Can I used the node.sdk in a react app? if so you do you have any samples?

erossignon, 2021-05-08:
NodeOPCUA is a  pure typescript/javascript nodejs technology and doesn’t rely on any binary node modules.  Whereas Node-opcua client and servers are meant to be used in the backend as a nodejs application as they require TCP Socket and a file system to manage the certificates PKI, most of the node-opcua modules can be reused in a web application and run on the browser side.

In fact, NodeOPCUA is developed in Typescript/Javascript. It already provides all the services and data structures for developing OPCUA applications. This is a great starting point for developing rich OPCUA front-end interfaces. All OPCUA data structures are readily JSON compatible can also be easily encoding/decoded into binary format.

danhellem, 2021-05-10:
thanks @erossignon 

interesting. My package.json is using 

```
"node-opcua": "^2.41.1",
"node-opcua-pki": "^2.9.8",
```

and then I have simple view in react like this...

```
import { OPCUAClient } from "node-opcua";

...

async main() {    
    let client = OPCUAClient.create({
      applicationName: "MyClient",
      connectionStrategy: {  initialDelay: 1000,
        maxRetry: 1},
      securityMode: 1,
      securityPolicy: "http://opcfoundation.org/UA/SecurityPolicy#None",
      endpoint_must_exist: false});

    await client.connect(endpointUrl);
    console.log("connected !");
  }
```

and I get the following error

[image]

I can follow the same code when creating a node server app and it works just fine.

Thoughts?

erossignon, 2021-05-10:
Sorry if it was not clear in my previous answer: OPCUAClient and OPCUAServer cannot work directly in the browser because of the need for TCP/Socket / Filesystem and crypto.  Most of the other modules could.

Sterfive has a (private) package that exposes the OPCUA Client API into a browser, this proxy module works in conjunction with a backend service that performs the actual connection on behalf of the client?  This module provides the exact same API as the nodejs OPCUAClient and marshals the call using HTTPS and WebSocket. This package is not open-source (yet).

khelkun, 2024-03-20:
For anyone struggling with this, a fast go is to use the api of your Next.js app from which you can use the node-opcua package to create your client and open a session. Then it's pretty straightforward to add simple routes to the api of the Next.js app that use the OPCUA session. Obviously it would require work to implement some meaningful RESTful API and eventually a websocket server with OPCUA subscriptions.

GoetzGoerisch, 2024-03-20:
Please use the REST API of OPC UA instead.

erossignon, 2024-03-20:
Alternatively use the Sterfive's NodeOPCUA WebProxy for that very purpose. 
This (commercial) module, built on top of NodeOPCUA by the author of NodeOPCUA (and writer of this answer), allows you to run your client NodeOPCUA code from within the browser and can address existing OPCUA Servers with classic opc.tcp:// transport protocol. ( see https://www.sterfive.com/product/premium-packages/ )