node-opcua/node-opcua #998: Cannot get node's data type

aljoshakoecher, 2021-05-17:
Hey everyone, I'm having trouble to get the data type of a node whose NodeId I get from resolving a NodeId string.

**Current behavior**
This is what I'm trying to do:
I start with a node ID string like `"ns=4;s=|var|CODESYS Control Win V3 x64.Application.PLC_PRG.calculator1.a"`. After resolving it to a proper NodeId object, I try to get the node's data type before later trying to write a value to this node:

```TypeScript
// I have a nodeIdString like the one above
const nodeIdString = "ns=4;s=|var|CODESYS Control Win V3 x64.Application.PLC_PRG.calculator1.a"

// I then get a NodeId object and this object looks okay, identifierType, namespace and value are set
const nodeId = NodeId.resolveNodeId(nodeIdString);

// I would then like to get this node's data type but this line throws an error: 
// Error: Invalid NodeId Identifier type => Numeric expected
const nodeType = await this.uaSession.getBuiltInDataType(nodeId);
```

**Describe the bug**
When trying to get the node's data type, I get an error:
Error: Invalid NodeId Identifier type => Numeric expected

**Expected behavior**
I expected to just get the node's data type, I cant understand why a Numeric NodeId Identifier type is expected here.

Any help is greatly appreciated!

**Context**

- (x ) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x ) I have installed node-opcua as a package ( using npm install )

- Description of the other OPCUA system I am trying to connect to:

    - Name: CODESYS Soft-PLC
    - Version: 3 (x64)

- **node-opcua version:** : 2.34.0
  
  

- **Node:**
  `node --version` = 12.16.2

erossignon, 2021-05-18:
> Error: Invalid NodeId Identifier type => Numeric expected

Can you provide longer stack trace ?

aljoshakoecher, 2021-05-18:
Hey @erossignon! Yeah sure, sorry that I forgot it in the first place. I updated my initial comment.

erossignon, 2021-05-22:
Shall be fixed in node-opcua@2.43.0 please confirm.

aljoshakoecher, 2021-05-22:
Awesome, that was quick. And it works 👍. Thank you very much.
But now I have a problem which might be related... 
Now I'm able to write to the node that caused the problem above. But I afterwards want to write to a node which has a custom DataType (an enumeration) as its Node Type. On this node, `getBuiltinDataType()` fails with:
```
Error: findBuiltInType : expecting a string undefined
    at Object.assert (C:\<path>\node_modules\node-opcua-assert\dist\index.js:13:21)
    at Object.findBuiltInType (C:\<path>\node_modules\node-opcua-factory\dist\factories_builtin_types.js:247:25)
    at get_encoder (C:\<path>\node_modules\node-opcua-variant\dist\variant.js:397:41)
    at encodeVariant (C:\<path>\node_modules\node-opcua-variant\dist\variant.js:191:24)
    at encodeVariant (C:\<path>\node_modules\node-opcua-variant\dist\variant.js:192:9)
    at Variant.encode (C:\<path>\backend\node_modules\node-opcua-variant\dist\variant.js:87:9)
    at Object.encodeDataValue (C:<path>\backend\node_modules\node-opcua-data-value\dist\datavalue.js:58:25)
    at WriteValue.encode (C:\<path>\node_modules\node-opcua-types\dist\_generated_opcua_types.js:23514:33)
    at C:\<path>\node_modules\node-opcua-types\dist\_generated_opcua_types.js:23583:97
    at Object.encodeArray (C:\<path>\node_modules\node-opcua-basic-types\dist\array.js:22:9)
```

Could this be related or are custom data types somehow not supported when using `getBuiltinDataType()`?

erossignon, 2021-11-27:
this is fixed in node-opcua@2.60.0 onward;
Please confirm

closed 2022-02-27 by erossignon