node-opcua/node-opcua #1000: Writing to a variable of MultiStateDiscreteType leads to unhandled exception in Node-opcua server

Graos, 2021-05-21:
**Describe the bug**

Writing to a variable of MultiStateDiscreteType from UAExpert client leads to unhandled exception in Node-opcua server

Steps to reproduce the behavior:

Load the Nodesetfile with a variable of MultiStateDiscreteType in the server with Read/Write access.
Start UAExpert and connect to the server.
Add the variable to the subscription and change the value of the variable.( node-opcua server gets into exception state)

**Screenshots**
[image]

(node:16452) UnhandledPromiseRejectionWarning: Error
    at Object.assert (\nodeopcuaServer\node_modules\node-opcua-assert\dist\index.js:13:21)
    at UAMultiStateDiscrete.checkVariantCompatibility (\nodeopcuaServer\node_modules\node-opcua-address-space\dist\src\data_access\ua_multistate_discrete.js:57:33)
    at UAMultiStateDiscrete.writeValue (\nodeopcuaServer\node_modules\node-opcua-address-space\dist\src\ua_variable.js:524:33)
    at UAMultiStateDiscrete.writeValue (\nodeopcuaServer\node_modules\thenify\index.js:65:46)
    at UAMultiStateDiscrete.writeAttribute (\nodeopcuaServer\node_modules\node-opcua-address-space\dist\src\ua_variable.js:610:22)
    at UAMultiStateDiscrete.writeAttribute (\nodeopcuaServer\node_modules\thenify\index.js:65:46)
    at ServerEngine.writeSingleNode (\nodeopcuaServer\node_modules\node-opcua-server\dist\server_engine.js:1039:17)
    at performWrite (\nodeopcuaServer\node_modules\node-opcua-server\dist\server_engine.js:1059:18)
    at \nodeopcuaServer\node_modules\async\dist\async.js:243:13
    at eachOfArrayLike (\nodeopcuaServer\node_modules\async\dist\async.js:500:13)
(Use `node --trace-warnings ...` to show where the warning was created)
(node:16452) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:16452) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua as a package ( using npm install )

- Device: ______
- OS version: ______

    - ( ) Windows : version : ______
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: ____UaExpert__
    - Version: ____1.5.1__
    - Manufacturer/Software vendor: _UA_____
    - link to the manufacturer or product site: https://

- **node-opcua version:** : 2.41.0
  
  

- **Node:**
  `node --version` =14.16.0

Graos, 2021-05-22:
Data type of the Value property of MultiStateDiscreteType variable  was defined as UINT6. When changed to UINT32. The issue was resolved. closing the case.

closed 2021-05-22 by Graos

erossignon, 2021-05-22:
well done ! you've earned the badge of the `1000th issue opener` ! :-)