node-opcua/node-opcua #1002: Allow overriding default transportTimeout

maximivanov, 2021-05-25:
**Is your feature request related to a problem? Please describe.**

Currently when trying to connect to a non-existing endpoint (e.g. when server port is mistyped), client blocks for 60 seconds before returning the error message. For certain use cases, we want faster feedback for the user. Unless I'm missing something, this timeout is not configurable at the moment. Feature request is to make it so.

**Describe the solution you'd like**

Allow passing `transportTimeout` as an option in the client constructor:

```js
const client = OPCUAClient.create({
  ...
  transportTimeout: 5 * 1000, // 5 seconds
}
```

**Describe alternatives you've considered**

```
connectionStrategy: {
  initialDelay: X,
  maxRetry: Y,
  maxDelay: Z
}
```

controls the timing **between** the connection requests, not the connection timeout itself.

**Additional context**

I can see transportTimeout is configured in ClientSecureChannelLayer: https://github.com/node-opcua/node-opcua/blob/b1296b54bdc5ef8c452abd851806f90816bda80e/packages/node-opcua-secure-channel/source/client/client_secure_channel_layer.ts#L388

In the client, secureChannel is instantiated without passing transportTimeout: https://github.com/node-opcua/node-opcua/blob/b1296b54bdc5ef8c452abd851806f90816bda80e/packages/node-opcua-client/source/private/client_base_impl.ts#L613

Thus there's no way to pass the transportTimeout when instantiating the client.

maximivanov, 2021-05-25:
Before creating this issue I searched existing issues by `transportTimeout` and didn't find anything. Only after creating it I found this related issue: #540. There's no solution there so this issue is still valid.

erossignon, 2023-05-07:
Now available in node-opcua@2.103.0

closed 2023-05-07 by erossignon