node-opcua/node-opcua #1004: node-opcua client gets BadResponseTooLarge, but UAExperte works ok

jonuriarte, 2021-05-26:
**Current behavior**

When connecting against a Siemens Simit OPC UA Server (v10.2 or v10.2 Update 1), client throws an error: _serviceResult = BadResponseTooLarge (0x80b90000)_. UA Expert (v 1.5.1 331) connects without a problem.

**Describe the bug**

Here is attached a zip file with Wireshark captures using UAExpert and node-opcua. 
Using node-opcua, when client gets **GetEndpointsResponse** message it sends a **CreateSessionRequest** message.  
When UAExpert gets **GetEndpointsResponse** message, it sends a **CloseSecureChannelRequest**. Then it restarts communication from the beginning, but this time UAExpert does not send **GetEndpointsRequest** message, going straight to creating session.

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

```
const { OPCUAClient } = require("node-opcua-client");
const {
    MessageSecurityMode,
    SecurityPolicy,
    AttributeIds,
    makeBrowsePath,
    ClientSubscription,
    TimestampsToReturn,
    MonitoringParametersOptions,
    ReadValueIdLike,
    ClientMonitoredItem,
    DataValue
  } = require("node-opcua");
  
const connectionStrategy = {
    initialDelay: 1000,
    maxRetry: 1
};
console.log("BEFORE OPCUAClient.create");
const client = OPCUAClient.create({
    connectionStrategy: connectionStrategy,
    securityMode: MessageSecurityMode.None,
    securityPolicy: SecurityPolicy.None,
    endpoint_must_exist: false
});
console.log("AFTER OPCUAClient.create");
const endpointUrl = "opc.tcp://192.168.103.25:47144";
  
(async () => {
    try {
        // step 1 : connect to
        console.log("BEFORE client.connect(endpointUrl);");
        await client.connect(endpointUrl);
        console.log("connected !");
    
        // step 2 : createSession
        console.log("BEFORE client.createSession();");
        const session = await client.createSession();
        console.log("session created !");
    
        // step 3 : browse
        const browseResult = await session.browse("RootFolder");
        
        console.log("references of RootFolder :");
        for (const reference of browseResult.references) {
            console.log("   -> ", reference.browseName.toString());
        }
        
        await timeout(10000);
            
        // step 6: finding the nodeId of a node by Browse name
        const browsePath = makeBrowsePath(
            "RootFolder",
            "/Objects/Server.ServerStatus.BuildInfo.ProductName"
        );
        
        const result = await session.translateBrowsePath(browsePath);
        const productNameNodeId = result.targets[0].targetId;
        console.log(" Product Name nodeId = ", productNameNodeId.toString());
    
        // close session
        await session.close();
    
        // disconnecting
        await client.disconnect();
        console.log("done !");
    } 
    catch(err) {
        console.log("An error has occured : ",err);
    }
})();

```
**Expected behavior**

**Screenshots**
**Desired script output**
[image]

**Actual script output**
[image]

**Gist**

 
gist:

**Context**

- ( X ) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- ( X ) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: ______

    - ( X ) Windows : version : W10 1809 (17763.1935)
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: Simit
    - Version: 10.2 / 10.2 Update 1
    - Manufacturer/Software vendor: Siemens
    - link to the manufacturer or product site: [https://new.siemens.com/global/en/products/automation/industry-software/simit.html](https://new.siemens.com/global/en/products/automation/industry-software/simit.html)
    - Demo version download is free, but a siemens account is needed.

- **node-opcua version:** :
   From package.json: "node-opcua": "^2.43.0"
  
  

- **Node:**
  `node --version` = v14.16.0

simonbowen, 2021-06-09:
I am also having this issue, did you get to the bottom of it?

AndreasHeine, 2021-06-09:
can you check with wireshark if the negotiated MaxMessageSize in HELLO/ACK is greater/smaller then the request which makes the fault?

jonuriarte, 2021-06-10:
HELLO message from node-opcua:

[image]

HELLO message from UAExpert:

[image]

Messages between node-opcua (192.168.234.129) and Simit (192.168.103.28):

[image]

Messages between UAExpert (192.168.234.129) and Simit (192.168.103.28):

[image]

[TEST_Wireshark_Simit_v10.2.zip](https://github.com/node-opcua/node-opcua/files/6631422/TEST_Wireshark_Simit_v10.2.zip)

> I am also having this issue, did you get to the bottom of it?

I updated to Simit v10.2 Update 2 and my node-opcua client connects perfectly. But as UAExpert had no problem connecting I think it may be interesting to try to find the problem and make node-opcua even better.

jonuriarte, 2021-11-10:
I've tried with version 2.57.0, but I've got the same problem.

erossignon, 2023-05-03:
fixed in [node-opcua@2.89.0 onward](https://github.com/node-opcua/node-opcua/releases/tag/v2.89.0)

closed 2023-05-03 by erossignon