node-opcua/node-opcua #1005: CVE-2021-23386: out-dated dns-packet dependency via bonjour

zenzxjul, 2021-05-26:
[CVE-2021-23386](https://nvd.nist.gov/vuln/detail/CVE-2021-23386)

The bonjour dependency uses multicast-dns, which in turn relies on a version of dns-packet that is vulnerable to CVE-2021-23386: see [issue 63 @ bonjour](https://github.com/watson/bonjour/issues/63)
There's already a [PR to upgrade the multicast-dns dependency within bonjour](https://github.com/watson/bonjour/pull/64), which updates the vulnerable dns-packet depedency.

erossignon, 2021-05-26:
Thank you for notifying us. We regularly keep up to date with the most up-to-date version of external dependencies; We will integrate this new version of bonjour with the fixed version of dns-packet module as soon as it will be released.

erossignon, 2021-11-19:
the patch version of dns-packet is now installed by bonjour 3.50
(https://github.com/mafintosh/dns-packet/issues/65) 

````
node-opcua-mono-repo@2.9.1 C:\NodeOPCUA\node-opcua
└─┬ bonjour@3.5.0 -> .\node_modules\.pnpm\bonjour@3.5.0\node_modules\bonjour
  └─┬ multicast-dns@6.2.3 -> .\node_modules\.pnpm\multicast-dns@6.2.3\node_modules\multicast-dns
    └── dns-packet@1.3.4 -> .\node_modules\.pnpm\dns-packet@1.3.4\node_modules\dns-packet
````

closed 2021-11-19 by erossignon