node-opcua/node-opcua #1007: readVariableValue deprecated

TomaszPrazniewski89, 2021-05-27:
Hi guys i have some problems when preparing a client.
I have made a connection to the endpointURL (        await client.connect(endpointUrl); says connected)

when trying to create a session 

i get such an error
connected !
An error has occured :  Error: Cannot find ANONYMOUS user token policy in end point description
    at createAnonymousIdentityToken (C:\Users\prazniewskit\Desktop\Siemens\webapp2\node_modules\node-opcua-client\source\private\opcua_client_impl.ts:117:15)
    at OPCUAClientImpl.createUserIdentityToken (C:\Users\prazniewskit\Desktop\Siemens\webapp2\node_modules\node-opcua-client\source\private\opcua_client_impl.ts:1157:41)
    at OPCUAClientImpl._activateSession (C:\Users\prazniewskit\Desktop\Siemens\webapp2\node_modules\node-opcua-client\source\private\opcua_client_impl.ts:875:14)
    at C:\Users\prazniewskit\Desktop\Siemens\webapp2\node_modules\node-opcua-client\source\private\opcua_client_impl.ts:393:22
    at ClientSecureChannelLayer.<anonymous> (C:\Users\prazniewskit\Desktop\Siemens\webapp2\node_modules\node-opcua-client\source\private\opcua_client_impl.ts:839:13)
    at modified_callback (C:\Users\prazniewskit\Desktop\Siemens\webapp2\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:1343:31)
    at process_request_callback (C:\Users\prazniewskit\Desktop\Siemens\webapp2\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:131:5)
    at ClientSecureChannelLayer._on_message_received (C:\Users\prazniewskit\Desktop\Siemens\webapp2\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:787:9)
    at MessageBuilder.<anonymous> (C:\Users\prazniewskit\Desktop\Siemens\webapp2\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:355:22)
    
    
    
    what even more when i try to read a value  using:
const dataValue2 = await session.readVariableValue("ns=3;s=Scalar_Simulation_Double");

it says readVariableValue deprecated

please help ;D

erossignon, 2021-05-27:
simply replace ` await session.readVariableValue("ns=3;s=Scalar_Simulation_Double");`
with
` await session.read({ nodeId: "ns=3;s=Scalar_Simulation_Double", attributeIds: AttributeIds.Value});`

erossignon, 2021-05-27:
an d replace 
`const session = await client.createSession({ userName: "user1", password: "password1" });`

with 
```javascript
 const session = await client.createSession({ 
     type: UserTokenType.UserName,
     userName: "user1", 
     password: "password1" 
});
```

TomaszPrazniewski89, 2021-05-27:
import {
    OPCUAClient,
    MessageSecurityMode,
    SecurityPolicy,
    AttributeIds,
    makeBrowsePath,
    ClientSubscription,
    TimestampsToReturn,
    MonitoringParametersOptions,
    ReadValueIdOptions,
    ClientMonitoredItem,
    DataValue,
    ReadValueId
} from "node-opcua";

const connectionStrategy = {
    initialDelay: 1000,
    maxRetry: 1
}

const options = {
    applicationName: "MyClient",
    connectionStrategy: connectionStrategy,
    securityMode: MessageSecurityMode.None,
    securityPolicy: SecurityPolicy.None,
    endpoint_must_exist: false,
};

const client = OPCUAClient.create(options);
// const endpointUrl = "opc.tcp://opcuademo.sterfive.com:26543";
const endpointUrl = "opc.tcp://192.168.0.14:55105";

async function main() {
    try {
        await client.connect(endpointUrl);
        console.log("connected !");

        const session = await client.createSession({
            type: UserTokenType.UserName,
            userName: "user1",
            password: "password1"
        });

        const dataValue = await session.read({
            nodeId: "ns=3;s=Scalar_Simulation_Double",
            attributeId: AttributeIds.Value
        });

        await session.close();

        await client.disconnect();
        console.log("done !");
    } catch (err) {
        console.log("An error has occured : ", err);
    }
}
main();

"type: UserTokenType.UserName," : is still highlited red

erossignon, 2021-05-27:
you need to import it

TomaszPrazniewski89, 2021-05-27:
May you know how to read a value from awaria elektryczna ( included in the
photo)?

TomaszPrazniewski89, 2021-05-27:
> It works yehha .. read and write as well
>
> On Thu, May 27, 2021 at 1:50 PM Tomasz Prazniewski <

May you know if its possible to connect to PLC directly?

erossignon, 2021-05-30:
Great! Closing then

closed 2021-05-30 by erossignon

josegoyo, 2022-09-26:
Hello, 

How can read multiple values using  into a for loop using  `session.read`, actually i have an array where are my tags, and i want to do a read using that array.

AndreasHeine, 2022-09-27:
@josegoyo yes you can read multiple nodes in one network request! as long its an Array of "readvalueidoptions" -> https://node-opcua.github.io/api_doc/2.32.0/interfaces/node_opcua.readvalueidoptions.html

https://node-opcua.github.io/api_doc/2.32.0/interfaces/node_opcua.clientsession.html#read
[image]

josegoyo, 2022-09-27:
Yes I write a function to read all tags using a GET request in my Rest API with express: 

```
const opcuaReadController = async (io) => {
    tags.forEach(async function (id) {
        const maxAge = 0;
        const nodeToRead = {
            nodeId: `ns=3;s="GOYO".${id}`,
            attributeId: AttributeIds.Value,
        };
        const dataValue = await session.read(nodeToRead, maxAge);
        console.log(" Data read =>", id, dataValue.value.value);

        let val = id.replaceAll('"', "");
        io.sockets.emit(val, {
            value: dataValue.value.value,
        });
    });
};
```
That worked for me.
You can ignore my socket emmit.

erossignon, 2022-09-27:
a preferred approach would be to group all reads into a single transaction the server:

```
const opcuaReadController = async (io) => {
    const nodesToRead = tags.map((id)=>({
         nodeId: `ns=3;s="GOYO".${id}`,
         attributeId: AttributeIds.Value,
    }));

   const dataValues = await session.read(nodesToRead, maxAge);
  
   dataValues.forEach((dataValue,index)=>{
         const id = tags[index]
         const val = id.replaceAll('"', "");
         io.sockets.emit(val, {
              value: dataValue.value.value,
         });
   });

};
```

The [Node OPCUA by example book (2022 edition)](https://leanpub.com/node-opcuabyexample-edition2022) will give you other best practices...