node-opcua/node-opcua #1009: Is openssl a runtime requirement?

maximivanov, 2021-05-28:
**Is your feature request related to a problem? Please describe.**

node-opcua client is configured to use an existing client certificate (meaning it shouldn't need to generate one):

```js
const client = OPCUAClient.create({
    ...
    certificateFile: '/path/to/cert',
    privateKeyFile: '/path/to/key'
})
```

Connection to a server fails if there's no `openssl` installed on the system.

Error:

```text
Command failed: "openssl" rand -out "/root/.config/node-opcua-default-nodejs/random.rnd" -hex 256
/bin/sh: openssl: not found
```

**Describe the solution you'd like**

I'm not sure if openssl is a runtime dependency. If it is, please consider adding a note in the readme.

Thanks!

erossignon, 2021-05-29:
Yes! openssl is a requirement. Certificates have to be regenerated to reflect the name of the computer, and dedicated private/public key need to be generated.
On Windows, a version of openssl is automatically downloaded and installed if not present. On Linux, it need to be installed:!
For instance on ubuntu/debian:
```
apt install openssl
```

maximivanov, 2021-05-29:
I see. Please consider updating the readme file 🙏

I may be missing something, but if I generate a certificate manually and supply it to node-opcua client, why does the client need to regenerate it? On the server, I can see it uses the certificate I generated (and not some other automatically generated one). 

What exactly openssl is used for in this case? Thanks.

erossignon, 2021-05-30:
We probably never came across a case where installing OpenSSL was a problem ;-)

You can circumvent this by preinstalling your private key in the following file:

       ~/.config/node-opcua-default-nodejs/PKI/own/private/private_key.pem

( with the right permission settings)

In this case, you don't even need to pass the privateKeyFile parameter to the OPCUAClient.create factory.

I guess you are using your client inside a  docker container right ?

In this case it is a good practice to store the ~/.config/node-opcua-default-nodejs/ in a persistant volume.

maximivanov, 2021-05-30:
Thanks for updating the documentation!

> I guess you are using your client inside a docker container, right ?

That's correct. The container is kept minimal so if we can avoid installing `openssl`, that'd be great.

I'll try the default cert path approach in the next few days and update this thread.

erossignon, 2021-05-30:
There is no little gain 😃  Openssl is about 600k on alpine. https://pkgs.alpinelinux.org/package/edge/main/x86/openssl.
Alternatively if you install openssl, node-opcua will create a default key pair for you, at first run. 
 If make the config folder persitent, the key pair will be reused when you restart your container.

maximivanov, 2021-06-01:
> There is no little gain 😃 Openssl is about 600k on alpine. https://pkgs.alpinelinux.org/package/edge/main/x86/openssl.

I agree it's not much. When I start thinking about the size, security and upgrades - best dependency is no dependency 😀

> Alternatively if you install openssl, node-opcua will create a default key pair for you, at first run.

We have an application running on multiple instances that come and go dynamically and originally each instance would have its own certificate auto-generated. Those would need to be whitelisted at the server. Having a pre-generated certificate shared among instances was a fix for that. From what I understand the ultimate solution should be maintaining a CA that is used to sign all clients certificates. Server trust the CA => client certificates can be rotated seamlessly. Having said that I'm still not sure whether there's any security implication to generating certificates at runtime vs having them pre-generated.

> You can circumvent this by preinstalling your private key in the following file:
   ~/.config/node-opcua-default-nodejs/PKI/own/private/private_key.pem

I've just tested locally on Mac, and the `node-opcua-default-nodejs` folder is not in the home folder but in `~/Library/Preferences`:

```bash
~/Library/Preferences/node-opcua-default-nodejs ❯ ls -la
total 0
drwxr-xr-x    6 maxivanov  staff   192 Jun  1 12:10 .
drwx------+ 207 maxivanov  staff  6624 Jun  1 12:09 ..
drwxr-xr-x    4 maxivanov  staff   128 Jun  1 12:09 issuers
drwxr-xr-x    5 maxivanov  staff   160 Jun  1 12:09 own
drwxr-xr-x    2 maxivanov  staff    64 Jun  1 12:09 rejected
drwxr-xr-x    4 maxivanov  staff   128 Jun  1 12:09 trusted
```

Here I got a bit confused. When I originally checked the contents of this folder there was a `PKI` folder in it as well as `issuers`, `own`, `rejected`, `trusted`. That is probably an artifact from the time I installed `2.41.1` couple months ago.

I then upgraded the client to version `2.44.0`, removed `~/Library/Preferences/node-opcua-default-nodejs`, and made a request. `~/Library/Preferences/node-opcua-default-nodejs` was recreated and the structure is what you see in the listing above.

So what's the correct path to the PKI folder? If you can reference where it's defined in the code that'd be much appreciated!

maximivanov, 2021-06-02:
I figured difference between PKI folder paths has to do with whether a default client certificate manager is used or not (I use custom). Also paths are platform-dependent (via [env-paths](https://www.npmjs.com/package/env-paths)). 

The bulk of code that is related is in `packages/node-opcua-certificate-manager/source/certificate_manager.ts` and `node-opcua-pki/lib/pki/certificate_manager.ts`.

I'm yet to confirm the fix worked in the container environment, will post an update here soon.

erossignon, 2021-06-02:
Hi Max, as you figured out the place where the "env-path" modules store the config data. It depends on the platform.

You can control the exact location of where to store the pki , by instantiating a OPCUACertificateManager prior to create the OPCUAClient object and passing it to the constructor paramters.

as in 

```javascript
const clientCertificateManager = new OPCUACertificateManager({
    location: "my_folder",
});
const client = OPCUAClient.create({
   ...
   clientCertificateManager
});

```

maximivanov, 2021-06-03:
I ended up reading the default location from the instantiated object:

```js
const clientCertificateManager = new OPCUACertificateManager({
    ...
});

const privateKeyPath = clientCertificateManager.privateKey;
const certificatePath = path.join( // I don't think I can get it computed, so preparing manually
    clientCertificateManager.rootDir, 
    'own/certs/client_certificate.pem' 
);

// save to privateKeyPath and certificatePath...

const client = OPCUAClient.create({
   ...
   clientCertificateManager
});
```

This works. Closing the issue. Thanks Etienne.

closed 2021-06-03 by maximivanov