node-opcua/node-opcua #1021: Server certificate is not accepted even when added to Trusted

maximivanov, 2021-06-30:
**Current behavior**

When connecting to a Mettler Toledo ACI400 Gateway and using security mode `Basic256Sha256`, nodeopcua client fails to verify server certificate. 

I've provided as much details as I could, but let me know if you need anything else. Thanks!

**Describe the bug**

Certificate is added to the "trusted" folder, yet it's not accepted.

Identical connection options used with UA Expert desktop client work just fine.

**Step by step instruction to reproduce**

```js
import {
  OPCUAClient,
  MessageSecurityMode,
  SecurityPolicy,
} from 'node-opcua-client'

const endpoint = 'opc.tcp://host:4840'

;(async () => {
  const client = OPCUAClient.create({
    securityMode: MessageSecurityMode.SignAndEncrypt,
    securityPolicy: SecurityPolicy.Basic256Sha256,
    endpointMustExist: false,
  })

  console.log('connecting...')

  try {
    await client.connect(endpoint)
  } catch (err) {
    console.log('client.connect error', err)
    process.exit(1)
  }

  console.log('connected')
})()
```

...it errors with the following error. Note the weird part `Certificate verification failed with err Good (0x00000)`. It sounds like there's no error?

```bash
❯ node test-mt-connection-issue-for-github-repo.mjs
connecting... 2021-06-30T23:04:49.664Z
23:04:50.468Z :client_base_impl              :164   serverCertificate =  c1c24aa61996b334f299fcb8a83f99ec0794b033
23:04:50.468Z :client_base_impl              :165   serverCertificate =  MIID0jCCArqgAwIBAgIIGwUcLb4T488wDQYJKoZIhvcNAQELBQAwJDEiMCAGA1UEAwwZTWV0dGxlciBUb2xlZG8gT1BDIFNlcnZlcjAeFw0yMTA2MjcxNDEwNDhaFw0yMjA2MjcxNDEwNDhaMCQxIjAgBgNVBAMMGU1ldHRsZXIgVG9sZWRvIE9QQyBTZXJ2ZXIwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQD41B5a9K01Hjzm5VDwP8v9qFY2ZsK7PCjbABO20LIw2ZpOWL9l8sQMoIs3WTNu9s7kf5xHYGCYXzmvMBp5skF7xtegzMFATtoWZ+2PG5dA2QD8wsx1GB63IjmHkwowzTB7PjAth2LGw0fCVlllmSMSiT4hNHjQynBF9M+H0jCL1GEnxeWdi/p1tXyMx61LOGVBpkK1D+KLp1PtzjoYOxmdV9LuLgzIzQq2LmtJ5jS9JpVFMwcVhjimFRzwkwpsYfq6z7FgWoVNY1MZxo99hJCdQwx6DqdG09y+aSUN/TzCfjQZVNwJ+xql1eY9c0J7phiPgvfYmIloRX4UZBubcBCRAgMBAAGjggEGMIIBAjAdBgNVHQ4EFgQUh/Fa7IRFVqGZ31g90EZvyyRDqtMwEgYDVR0TAQH/BAgwBgEB/wIBADBTBgNVHSMETDBKgBSH8VrshEVWoZnfWD3QRm/LJEOq06EopCYwJDEiMCAGA1UEAwwZTWV0dGxlciBUb2xlZG8gT1BDIFNlcnZlcoIIGwUcLb4T488wDgYDVR0PAQH/BAQDAgL0MCAGA1UdJQEB/wQWMBQGCCsGAQUFBwMBBggrBgEFBQcDAjBGBgNVHREEPzA9hip1cm46ZGVza3RvcC1rbWtlNGNiOk1ldHRsZXJUb2xlZG9PUENTZXJ2ZXKCD2Rlc2t0b3Ata21rZTRjYjANBgkqhkiG9w0BAQsFAAOCAQEA1pSV9RbnHLGckEBA0VrVvZ1k+LsfLO0ElP1Py4yl9sK1+UCz7LRCoMkPrIXOf5q4165qMQLixYH1pw3j7hnOYURV8rl4d7mxjXUZTk/dAfUmy0TIiwIbCsy7BgOB3aYULg0oGuiDSGcBuBEtNHMNYkVGDiOIjekp1FhgPXBX1V9fGDjOwaOCWDuHK9XP1ztuJJxv143GRfeu1CgXFfv4rXVC6Ea4TSZooEB37JKIoZfX35nL+KEQBZj9afBQUOEHIs7l/t6qasAAcDtWPmPSuJZF6yPNa89dbr7UsjAS0vfq6X9vbz9oFaeDYjfdtWQb23UDxmcr88oCJmqVGwac3w==
23:04:50.468Z :client_base_impl              :1018  [NODE-OPCUA-W25] client's server certificate verification has failed  server Certificate verification failed with err Good (0x00000)
23:04:50.469Z :client_base_impl              :1019                    /Users/maxivanov/Library/Preferences/node-opcua-default-nodejs/PKI
23:04:50.469Z :client_base_impl              :1020                    MIID0jCCArqgAwIBAgIIGwUcLb4T488wDQYJKoZIhvcNAQELBQAwJDEiMCAGA1UEAwwZTWV0dGxlciBU
 ...                                                                 b2xlZG8gT1BDIFNlcnZlcjAeFw0yMTA2MjcxNDEwNDhaFw0yMjA2MjcxNDEwNDhaMCQxIjAgBgNVBAMM
 ...                                                                 GU1ldHRsZXIgVG9sZWRvIE9QQyBTZXJ2ZXIwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQD4
 ...                                                                 1B5a9K01Hjzm5VDwP8v9qFY2ZsK7PCjbABO20LIw2ZpOWL9l8sQMoIs3WTNu9s7kf5xHYGCYXzmvMBp5
 ...                                                                 skF7xtegzMFATtoWZ+2PG5dA2QD8wsx1GB63IjmHkwowzTB7PjAth2LGw0fCVlllmSMSiT4hNHjQynBF
 ...                                                                 9M+H0jCL1GEnxeWdi/p1tXyMx61LOGVBpkK1D+KLp1PtzjoYOxmdV9LuLgzIzQq2LmtJ5jS9JpVFMwcV
 ...                                                                 hjimFRzwkwpsYfq6z7FgWoVNY1MZxo99hJCdQwx6DqdG09y+aSUN/TzCfjQZVNwJ+xql1eY9c0J7phiP
 ...                                                                 gvfYmIloRX4UZBubcBCRAgMBAAGjggEGMIIBAjAdBgNVHQ4EFgQUh/Fa7IRFVqGZ31g90EZvyyRDqtMw
 ...                                                                 EgYDVR0TAQH/BAgwBgEB/wIBADBTBgNVHSMETDBKgBSH8VrshEVWoZnfWD3QRm/LJEOq06EopCYwJDEi
 ...                                                                 MCAGA1UEAwwZTWV0dGxlciBUb2xlZG8gT1BDIFNlcnZlcoIIGwUcLb4T488wDgYDVR0PAQH/BAQDAgL0
 ...                                                                 MCAGA1UdJQEB/wQWMBQGCCsGAQUFBwMBBggrBgEFBQcDAjBGBgNVHREEPzA9hip1cm46ZGVza3RvcC1r
 ...                                                                 bWtlNGNiOk1ldHRsZXJUb2xlZG9PUENTZXJ2ZXKCD2Rlc2t0b3Ata21rZTRjYjANBgkqhkiG9w0BAQsF
 ...                                                                 AAOCAQEA1pSV9RbnHLGckEBA0VrVvZ1k+LsfLO0ElP1Py4yl9sK1+UCz7LRCoMkPrIXOf5q4165qMQLi
 ...                                                                 xYH1pw3j7hnOYURV8rl4d7mxjXUZTk/dAfUmy0TIiwIbCsy7BgOB3aYULg0oGuiDSGcBuBEtNHMNYkVG
 ...                                                                 DiOIjekp1FhgPXBX1V9fGDjOwaOCWDuHK9XP1ztuJJxv143GRfeu1CgXFfv4rXVC6Ea4TSZooEB37JKI
 ...                                                                 oZfX35nL+KEQBZj9afBQUOEHIs7l/t6qasAAcDtWPmPSuJZF6yPNa89dbr7UsjAS0vfq6X9vbz9oFaeD
 ...                                                                 YjfdtWQb23UDxmcr88oCJmqVGwac3w==
23:04:50.470Z :client_base_impl              :1021                   verify that server certificate is trusted or that server certificate issuer's certificate is present in the issuer folder
[... 91 lines cut ...]
    - (x) MacOs : version : Big Sur 11.2.3
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: _Mettler Toledo ACI400 Gateway_____
    - Version: ______
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: https://www.mt.com/dam/product_organizations/industry/IndustrialTerminals/30576844_00_MAN_UG_ACI400_EN.pdf

- **node-opcua version:** :
  
  

```bash
> npm list --depth=0
node-opcua-poc@1.0.0 /Users/maxivanov/src/local/node-opcua-poc
├── node-opcua-certificate-manager@2.45.0
├── node-opcua-client@2.45.0
└── node-opcua-crypto@1.8.0
```

- **Node:**
  `node --version` = 14.16.1

erossignon, 2021-07-02:
the "with err Good" part of the error let me think that you have not updated the node-opcua-xxx modules to their required versions.

Please update node-opcua packages using

`npx npm-check-updates -u node-opcua*`

and re-run 
`npm install` or `yarn`

maximivanov, 2021-07-07:
It didn't help unfortunately. To double check I removed the `node_modules` folder altogether and installed from scratch. 

Getting the same error. Is there anything else I can do to help troubleshoot this?

erossignon, 2021-07-08:
Please contact support@sterfive.com for investigation of your specific case

closed 2022-03-27 by erossignon