node-opcua/node-opcua #1025: Hello guys I am trying to receive data from opcua server in javascript and i have this error on terminal of my raspberry pi

FabioAlmeida25, 2021-07-09:
[image]
[image]
[image]
[image]

AndreasHeine, 2021-07-09:
check your NodeId with UaExpert!
looks like a Siemens PLC here is a screenshot from my Test PLC:
[image]

FabioAlmeida25, 2021-07-09:
Hello @AndreasHeine for your help.
In my case don't apear that way and I don't know why.
[image]

AndreasHeine, 2021-07-09:
ah ok use -> "ns=4;i=4" in your case your "IdentifierType" is numeric not string!

FabioAlmeida25, 2021-07-09:
Error again:
[image]
[image]
[image]

FabioAlmeida25, 2021-07-09:
Ok I receive the value 10, the same value as my variável_1, so its ok. Sorry my english by the way.

FabioAlmeida25, 2021-07-09:
I haved tried that way, but when I opened my webpag my graph don't shown any value. And I know it's strange because I established connection between my webserver and  the client using python language. And I used in that case (ns=4;i=4). My problem now I think it is in the rest of the code.

[image]

closed 2021-07-09 by FabioAlmeida25