node-opcua/node-opcua #1028: [Client] If folder of certificateFile is readonly Client can't connect

bruegth, 2021-07-14:
**Current behavior**

Client can't connect to server if certificateFile Folder is read only.

**Describe the bug**

Node process hung at `client.connect(url)` 

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1- make directory where certificate file is stored read only
2- try to connect to any opcua server with certificateFile

**Expected behavior**

connect() with read-only certificate folder will work.

**Information**

node-opcua version 2.33 is working.

Maybe related to ".lock" file: https://github.com/node-opcua/node-opcua/blob/bc0fd185f4905cd03c83c7a278365424f2b46bac/packages/node-opcua-client/source/private/client_base_impl.ts#L749

**Context**

- (x ) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- OS version: 

    - ( ) Windows : version : ______
    - (x) Linux : version : docker image node:16
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- **node-opcua version:** :
2.47.0

- **Node:**
  `node --version` = v16.4.2

jvstein, 2021-11-17:
Thank you! I was running into this and having difficulty figuring out why the connect was stalling.

tuomotalvitie, 2024-03-14:
On related note: server initialization freezes without errors if there are missing write permissions under PKI.