node-opcua/node-opcua #1033: Error: getExtensionObjectConstructorFromBinaryEncoding cannot find constructor for binaryEncoding

priyanka01deb, 2021-07-28:
Hello,
I'm trying to read a Extension Object Type variable with Siemens S7-1500 PLC.
[image]

In the above code, I'm getting error.
Error: getExtensionObjectConstructorFromBinaryEncoding cannot find constructor for binaryEncoding ns=4;i=5003

But I am able to see the structure type in log 
[image]

My Address Model looks like below :
[image]

[image]

erossignon, 2021-08-05:
a fix will be available soon, ( stay tuned on #1037)

erossignon, 2021-08-17:
fixed in [node-opcua@2.49.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.49.0)

closed 2021-08-17 by erossignon

priyanka01deb, 2021-08-18:
Thank you so much for your support!

mg90707, 2021-09-30:
When I disable the "Legacy data type definitions according to OPC UA specification <= V1.03" on my Siemens 1511 PLC I get the following error when attempting to read a custom structure:
`....populate_data_type_manager_104:49    Error Error:  Cannot find dataType Definition ! with nodeId =ns=3;i=3400`
and it seems as if node-opc-ua (v2.51.0) attempts to read the old XML type defintion which is not present anymore.
When activating the 1.03 "TypeDictionary" again it works without problems. However with UA Expert I can browse the type definitions (V1.04). Do I have to set some flag or anything before node-opc-ua works with the new type defition format?

Thanks for your help!

david114, 2022-07-26:
@mg90707
Have you found a feasible solution to this problem?
I'm currently facing the same issue, receiving similar error messages, although setting 
`Legacy data type definitions according to OPC UA specification <= V1.03
`
to true.

```
| 11:54:58.249Z :populate_data_type_manager_104:38    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=6522
|  ...                                                    at /usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:31:23
|  ...                                                    at Generator.next (<anonymous>)
|  ...                                                    at fulfilled (/usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:5:58)  
|  ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
| 11:54:58.250Z :populate_data_type_manager_104:167   err= Error
|  ...                                                    at assert (/usr/src/app/node_modules/node-opcua-assert/dist/index.js:11:21)
|  ...                                                    at /usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:164:52
|  ...                                                    at Generator.next (<anonymous>)
|  ...                                                    at fulfilled (/usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:5:58)  
|  ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
| 11:54:58.311Z :populate_data_type_manager_104:38    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=6525
|  ...                                                    at /usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:31:23
|  ...                                                    at Generator.next (<anonymous>)
|  ...                                                    at fulfilled (/usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:5:58)  
|  ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
| 11:54:58.312Z :populate_data_type_manager_104:167   err= Error
|  ...                                                    at assert (/usr/src/app/node_modules/node-opcua-assert/dist/index.js:11:21)
|  ...                                                    at /usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:164:52
|  ...                                                    at Generator.next (<anonymous>)
|  ...                                                    at fulfilled (/usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:5:58)  
|  ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
| 11:54:58.359Z :populate_data_type_manager_104:38    Error Error:  Cannot find dataType Definition ! with nodeId =ns=3;i=3400
|  ...                                                    at /usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:31:23
|  ...                                                    at Generator.next (<anonymous>)
|  ...                                                    at fulfilled (/usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:5:58)  
|  ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
| 11:54:58.359Z :populate_data_type_manager_104:167   err= Error
|  ...                                                    at assert (/usr/src/app/node_modules/node-opcua-assert/dist/index.js:11:21)
|  ...                                                    at /usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:164:52
|  ...                                                    at Generator.next (<anonymous>)
|  ...                                                    at fulfilled (/usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:5:58)  
|  ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
| 11:54:58.745Z :populate_data_type_manager_104:38    Error Error:  Cannot find dataType Definition ! with nodeId =ns=2;i=15888
|  ...                                                    at /usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:31:23
|  ...                                                    at Generator.next (<anonymous>)
|  ...                                                    at fulfilled (/usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:5:58)  
[... cut ...]
|  ...                                                    at /usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:164:52
|  ...                                                    at Generator.next (<anonymous>)
|  ...                                                    at fulfilled (/usr/src/app/node_modules/node-opcua-client-dynamic-extension-object/dist/private/populate_data_type_manager_104.js:5:58)  
|  ...                                                    at processTicksAndRejections (node:internal/process/task_queues:96:5)
```