node-opcua/node-opcua #1038: Defining three dimension matrix variable leads to false warning in Nodeopcua server

Graos, 2021-08-06:
Defined a boolean type three dimension matrix variable in the namespace with dimension [4,4,4] i.e. total 64 elements.

This leads to warning in nodeopcua server  with the following print
Internal Error: matrix dimension doesn't match :  { /* DataValue */
 ...                                                   value: Variant(Matrix[ 4,4,4 ]<Boolean>, l= 64, value=[true,true,true,true,true,true,true,true,true,true,...])
 ...                                                   statusCode:      Good (0x00000)
 ...                                                   serverTimestamp: 2021-08-06T08:24:20.833Z $ 831.100.000
 ...                                                   sourceTimestamp: 2021-08-06T08:24:20.833Z $ 831.100.000

because of the false logic in UAVariable.js line 1265

       if (dataValue.value.arrayType === node_opcua_variant_1.VariantArrayType.Matrix) {
            if (dataValue.value.value.length !== 0 && dataValue.value.value.length !== dataValue.value.dimensions[0] * dataValue.value.dimensions[1]) {
                warningLog("Internal Error: matrix dimension doesn't match : ", dataValue.toString());
            }
        }

Length is compared based on only the first two dimensions size which is wrong 
dataValue.value.dimensions[0] * dataValue.value.dimensions[1]

Node-opcua version: 2.47.0

erossignon, 2021-08-23:
fixed in node-opcua@2.50.0, please confirm

Graos, 2021-08-25:
I can confirm that version 2.50.0 resolved the issue stated above. Many thanks for your time in resolving this.

closed 2021-08-25 by Graos