node-opcua/node-opcua #1041: ########################################### OPENSSL ERROR ########################################### CWD = . Command failed: "openssl" version /bin/sh: openssl: not found

jhonalcala3211, 2021-08-18:
**Current behavior**
Here's my sample code
const BPromise = require('bluebird')
const opcua = BPromise.promisifyAll(require('node-opcua'))
const OPCUAClient = opcua.OPCUAClient
const OPCUAManager = opcua.OPCUACertificateManager

const clientCertificateManager = new OPCUAManager()

  const privateKeyPath = clientCertificateManager.privateKey
  const certificatePath = path.join( // I don't think I can get it computed, so preparing manually
    clientCertificateManager.rootDir,
    'own/certs/client_certificate.pem'
  )
  
  const connectionStrategy = {
  initialDelay: 1000,
  maxRetry: 1
}

  const options = {
    connectionStrategy: connectionStrategy,
    keepSessionAlive: true,
    endpointMustExist: false,
    clientCertificateManager
  }

let client = OPCUAClient.create(options)
  client.connect('opc.tcp://20.98.172.75:4840')
    .then(() => {
      console.log('connected')
      })

########################################### OPENSSL ERROR ###########################################
CWD = .
Command failed: "openssl" version
/bin/sh: openssl: not found

Command failed: "openssl" rand  -out "/root/.config/node-opcua-default-nodejs/random.rnd" -hex 256
/bin/sh: openssl: not found

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. ....
2. ....
3. ....

**Expected behavior**

It should be connected to the Server.

**Gist**

 
gist:

**Context**

- (x ) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x ) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: ______
- OS version: ______

    - ( ) Windows : version : ______
    - (x ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: ______
    - Version: ______
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: https://

- **node-opcua version:** :
  
  

- **Node:**
  `node --version` =

jhonalcala3211, 2021-08-18:
node --version = 8.17.0

Need help about this one.

erossignon, 2021-08-23:
openssl must be installed on your system as a prerequisite.

erossignon, 2021-08-24:
duplicates #1009

closed 2021-08-24 by erossignon