node-opcua/node-opcua #1043: node-opcua-pki Flashing command prompts on windows

BartaG512, 2021-08-24:
-->

**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

I would like to make disappear the flashing popups each time the application starts.
This could be achieved by using options `windowsHide: true` in exec method
[Source nodejs docs](https://nodejs.org/api/child_process.html#child_process_child_process_exec_command_options_callback)
 
**Additional context**

Add any other context or screenshots about the feature request here.
Editing the two js files in dist folder i was able to stop the popups.
install_prerequisite.js
[image]
toolbox.js
[image]

erossignon, 2021-08-24:
issue could be registered at https://github.com/node-opcua/node-opcua-pki where the node-opcua-pki package in maintained.

erossignon, 2021-09-27:
fixed in node-opcua-pki@2.12.1

closed 2021-09-27 by erossignon