node-opcua/node-opcua #1050: Acknowledge Alarm error BadNodeIdUnknown

boualid, 2021-09-10:
**Current behavior**
Currently i am developing OPCUA client using Node Red. I am able to read the event/Alarm from OPC UA server.
Now i want to acknowledge the same, so i had create a data flow, with OPCUA nodes , that allow to call acknowledgement method for opcua.

**Describe the bug**
Although i insert the right parameters (conditionId, eventId and comment) i get below error message.
I take the values of the parameter from UA EXpert.
[image]

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. the below screenshot show the alarms that have been generated by the server
[image]

2. I want to acknowledge for example  the «Alm03», so 
2.1 the condtitionID == 2673
[image]
2.2 the eventId == 0x31b1dea5609b89ece2f0b856321fb58600000004
[image]
2.3 In the end i add comments

**Expected behavior**
i should be able to acknowledge the alarm «Alm03»

**Gist**

 
gist:

**Context**

- (X ) my request is related to node-opcua acting as an OPCUA CLIENT
- (X) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- ( ) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - (X) node-red
    - ( ) other : please specify

- Device: ______
- OS version: ______

    - ( ) Windows : version : ______
    - (X) Linux : version : Ubuntu 18.04.2 LTS(Bionic Beaver)
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- **node-opcua version:** :
  
  

- **Node:**
  `node --version` =v16.5.0

boualid, 2021-09-14:
my problem is resolved

closed 2021-09-14 by boualid

AcureroAdrian, 2023-11-24:
How was the problem solved?

AcureroAdrian, 2023-11-24:
@boualid How was the problem solved?