node-opcua/node-opcua #1051: Misleading prints "More than one HasChild reference have been found for parent of object"

Graos, 2021-09-15:
NODE-OPCUA Server version : 2.50.0
NODEJS version : v14.16.0

It is Ok to have UAVariable having a subtype reference (from HasChild ) referenced in more than one Object.

e.g 4:AllValveLockout UAVariable is referenced with reference Type "ns=2;i=1183" (HasInterLock which is subtype of HasChild) from multiple parent Objects (ns=4;i=4576,ns=4;i=4406)
[image]

Following are the prints thrown by the NODE_OPCUA server which has no severity contained in it

 More than one HasChild reference have been found for parent of object
    object node id: ns=5;i=24155 5:AllValveLockout
    browseResults:
  More than one HasChild reference have been found for parent of object
    object node id: ns=5;i=24155 5:AllValveLockout
    browseResults:
  More than one HasChild reference have been found for parent of object
    object node id: ns=5;i=24155 5:AllValveLockout
    browseResults:

erossignon, 2021-09-16:
It is not OK for one child to have more than one parent Node. 

The HasInterLock reference would need to be NonHiearchical reference, I guess; 
I would argue that the OPCUA Model defines the HasInterLock reference as a sub referenceof HasChild.
Can you share the model ?

Graos, 2021-09-17:
I have attached the MDIS companion nodeset file which is used in the compliance testing in OPC UA lab. HasInterlock is the subtype of HasChild and the same is added as child reference to multiple Parent Objects. see NodeId: ns=5;i=24155 5:AllValveLockout
[Instance.MDIS.NodeSet2.xml.zip](https://github.com/node-opcua/node-opcua/files/7182489/Instance.MDIS.NodeSet2.xml.zip)

erossignon, 2021-09-17:
The specs says:
https://reference.opcfoundation.org/v104/Core/docs/Part3/7.5/

>7.5 HasChild ReferenceType ToC Previous Next
>The HasChild ReferenceType is an abstract ReferenceType; only subtypes of it can be used. It is a subtype of HierarchicalReferences.
>The semantic is to indicate that References of this type span a non-looping hierarchy.
>Starting from Node “A” and only following References of the subtypes of the HasChild ReferenceType it shall never be possible to return to “A”. But it is allowed that following the References there may be more than one path leading to another Node “B”.

Therefore having 2 parents for one child would cause 2 path leading to the same child element, and this would break the HasChild Reference rules;

So, the exception is  valid and is here to detect such cases ... 

The instance file of the MDIS file is wrongly implements and shall be changed.

I understand that the "AllValveLockout" is not a component but rather a container that should loossly point to existing children object that are already attached to their respective parent.

In this case I would turn the AllValveLockout into a UAFolder and use the "Organises" Reference to back point to the corresponding referenced ValveLockout.

erossignon, 2021-09-17:
I've opened a ticket for the specification working group to suggest clarification in the specification regarding HasChild.
https://mantis.opcfoundation.org/view.php?id=7261

Graos, 2021-09-20:
This is the comment i received from the Experts in OPC UA Foundation

"The rule for haschild reference is that It can not get back to the same node following forward direction reference – it is perfectly legal to get to the same node from two or more starting points, It just can not have a loop.  I’m fairly sure that we do not have any loops – having the same interlock reference from more then one object is normal and expected in the MDIS model."

erossignon, 2021-09-21:
Correct; I changed the code to verify that the "Aggregates" reference  is not exposing 2 parents as opposed to the "HasChild" reference. No release date yet.

erossignon, 2021-09-27:
fixed in node-opcua@2.53.0 (next)

closed 2021-09-27 by erossignon