node-opcua/node-opcua #1057: Page Refresh

FabioAlmeida25, 2021-10-04:
Hello guys!

I am using this modules:

const express = require("express");  
const EventEmitter = require('events');    
const chalk = require("chalk");
//const path = require('path');
const socketIO = require("socket.io");
const port = 80;
const {
    AttributeIds,
    OPCUAClient,
    TimestampsToReturn,
    }= require("node-opcua");

When i refresh my webpage my values are gone, only show again if some value change. 

Example:

 ESTADO_SAIDA_P_SA.on("changed", (dataValue) => {
            console.log("SAIDA_P_SA = ", dataValue.value.toString());
            io.sockets.emit('SAIDA_P_SA', {
                value: dataValue.value.value,
            });
        });

What I want is when i refresh my webpag the values on opcua server, show on my webpag.

AndreasHeine, 2021-10-04:
the answer is simple you need to store that last valuechange somewhere! if you reload you can then render the last value into the html template and then connect to the ws in order to update the dom if the value changes.

FabioAlmeida25, 2021-10-08:
#

erossignon, 2022-08-10:
Closing due to lack of activity

closed 2022-08-10 by erossignon