node-opcua/node-opcua #1058: Publish on opcua server

FabioAlmeida25, 2021-10-04:
Hello guys! 
I am using node opc ua, and I know how to do a subscription on opcua server, but I don't know how to do a publish on my opcua server.
Anyone can help me please, or send me a example??

Thank You.

AndreasHeine, 2021-10-04:
https://reference.opcfoundation.org/v104/Core/docs/Part4/5.13.1/

a datachange notification is triggered if the vlaue changes or the statuscode!

FabioAlmeida25, 2021-10-05:
Thanks for your response.
For example, imagine I want to send number of button clicks to opcua server, how I do that?
        
io.on('connection', function(client) {  
            //when the server receives clicked message, do this
            client.on('clicked', function(data) {
                  clickCount++;
                  //send a message to ALL connected clients
                  io.emit('buttonUpdate', clickCount);
            });
        });

AndreasHeine, 2021-10-06:
Websocketclient (sends msg) -> ```[ Websocketserver (onmessage updates Variable1)  -> OPC UA Server (UAVariable: getter returns Variable1 value) ]``` -> OPC UA Client

AndreasHeine, 2021-10-09:
is your http backend opc ua server? or client?
i assume your events get lost because of open ends... its kind of hard to understand your code!

FabioAlmeida25, 2021-10-09:
I am learning from scratch, and it's the first time working with node opcua and socket.io. I am using for my master's thesis.
Sorry if it's hard for you to understand, in mid time I finally made it works.

html code:
[image]

app.js:
[image]

AndreasHeine, 2021-10-09:
[image]

Webserver + OPC UA Server:
```js
// Webserver Part

const express = require('express');
const app = express();

const http = require('http');
const server = http.createServer(app);

const { Server } = require("socket.io");
const io = new Server(server);

app.get('/', (req, res) => {  res.sendFile(__dirname + '/index.html');});

let variable1 = 0;

io.on('connection', (socket) => {  
    socket.emit("hello")
    console.log('a user connected');
    socket.on('buttonClick', (data) => {
        variable1 = variable1 + 1  
        console.log("buttonClick", variable1) 
    })
});

server.listen(3000, () => {  
    console.log('listening on *:3000');
});

// OPC UA Part -> Server

const { OPCUAServer, Variant, DataType, StatusCodes } = require("node-opcua");

(async () => {
    const server = new OPCUAServer({
        port: 4840,
        resourcePath: "/UA",
    });
    await server.initialize();
    console.log("initialized");
    const addressSpace = server.engine.addressSpace;
    const namespace = addressSpace.getOwnNamespace();
    
    namespace.addVariable({

        organizedBy: addressSpace.rootFolder.objects,

        nodeId: "s=clickcount", // a string nodeID
        browseName: "clickcount",
        dataType: "Int32",
        value: {
            get: () => new Variant({ dataType: DataType.Int32, value: variable1})
        }
    });

    server.start(function() {
        console.log("Server is now listening ... ( press CTRL+C to stop)");
        console.log("port ", server.endpoints[0].port);
        const endpointUrl = server.endpoints[0].endpointDescriptions()[0].endpointUrl;
        console.log(" the primary server endpoint url is ", endpointUrl);
    });
})();
```

index.html:
```html
<!DOCTYPE html>
<html lang="en">
<head>
    <meta charset="UTF-8">
    <meta http-equiv="X-UA-Compatible" content="IE=edge">
    <meta name="viewport" content="width=device-width, initial-scale=1.0">
    <title>Document</title>
</head>
<body>
    <button id="button1">Click Me!</button>

    <script src="/socket.io/socket.io.js"></script>
    <script>
        const socket = io("http://127.0.0.1:3000/");
        socket.on("connect", () => {
            console.log("connected")
            
            socket.on("hello", () => {
                console.log("hello from backend")
            })

            const button1 = document.getElementById('button1');
            button1.addEventListener('click', function(e) {    
                e.preventDefault(); 
                console.log("clicked")      
                socket.emit('buttonClick');      
            });
        })

    </script>
</body>
</html>
```

erossignon, 2023-05-07:
@FabioAlmeida25  finally got it working;  Closing

closed 2023-05-07 by erossignon