node-opcua/node-opcua #1059: Automatic reconnect does not restore subscription and related registered callbacks are not invoked

mg90707, 2021-10-05:
**Current behavior**

node-opcua fails to restore a subscription properly after a forced disconnect (pulling cable). Also many callbacks related to the reconnect process are not invoked making manually detecting the broken state very hard.

**Describe the bug**

I have a S7-1511 PLC with an OPC server running.  connect to this PLC via node-opcua and make a subscription + monitor some items. When the connection gets interrupted (pulling the network cable for ~20 seconds) node-opcua fails to restore the monitored items after the connection is established again. Also most of the registered callbacks on client and session related to the reconnect process are NOT invoked. Also the PLC shows that there were incorrect requests during the session restore. 

It seems like node-opcua realizes that the subscription is missing as the log contains
`WARNING :   SERVER TELLS THAT IT HAS NO SUBSCRIPTION , BUT CLIENT DISAGREE`.
But for some reason this message is also 5 times in the log when I only interrupted the connection once. The same goes for the following message in the log
` WARNING : SERVER TELLS THAT THE SESSION HAS CLOSED ...`

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. Start application and let it connect to the PLC
2. Wait a short time
3. Pull network cable for ~20 seconds (session is timed out on the PLC)
4. Plug the cable back in
5. Wait for node-opcua to reconnect
6. See that the session is improperly restored on the PLC diagnostics

**Expected behavior**

I'd expect node-opcua to restore the connection properly and also to fire off the registered callbacks regarding reconnection (e.g. client.o("start_reconnection"/"reconnection_attempt_has_failed"/"after_reconnection"...) etc.

**Screenshots**

This is the server diagnostics after the inital connect
[image]

this is after the connection was improperly restored (no subscription and no monitored item)
[image]

**Gist**

 
gist:

```
const {
    AttributeIds,
    OPCUAClient,
    TimestampsToReturn,
    MessageSecurityMode,
    SecurityPolicy
} = require("node-opcua");

const endpointUrl = "opc.tcp://192.168.137.101:4840";
const nodeId = 'ns=3;s="SIM"."TRUE"';

async function main() {
    try {
        // Create a client
        const connectionStrategy = {
            maxRetry: Number.MAX_SAFE_INTEGER
        };
        const client = OPCUAClient.create({
            applicationName: "Test",
            clientName: "Test",
            connectionStrategy,
            requestedSessionTimeout: 15000,
            endpointMustExist: false,
            keepSessionAlive: true,
            securityMode: MessageSecurityMode.None,
            securityPolicy: SecurityPolicy.None,
        });
        client
[... 722 lines cut ...]
06:27:51.006Z :client_session_keepalive_manag:112   ClientSessionKeepAliveManager#ping_server timeSinceLastServerContact= 10002 timeout 15000
06:27:51.006Z :client_secure_channel_layer   :917   Adjusted timeout =  60000
06:27:51.006Z :client_base_impl              :318     Client 0 Test emiting  send_request
06:27:51.007Z :client_base_impl              :318     Client 0 Test emiting  send_chunk
06:27:51.011Z :client_base_impl              :318     Client 0 Test emiting  receive_chunk
06:27:51.012Z :message_builder               :134    Sequence Header SequenceHeader { sequenceNumber: 85, requestId: 35 }
06:27:51.012Z :message_builder               :257   client0 Sequence Number =  85
06:27:51.012Z :message_builder               :201   client0 message size = 66        body size   = 42        requestHandle =  35 ReadResponse
06:27:51.013Z :client_secure_channel_layer   :924   ------------------------------------- Client receiving response  ReadRequest 35 ReadResponse err= null securityTokenId= 1
06:27:51.013Z :client_base_impl              :318     Client 0 Test emiting  receive_response
06:27:51.013Z :client_session_keepalive_manag:144   emit keepalive
Session keepalive received.
06:28:01.016Z :client_session_keepalive_manag:112   ClientSessionKeepAliveManager#ping_server timeSinceLastServerContact= 10003 timeout 15000
06:28:01.016Z :client_secure_channel_layer   :917   Adjusted timeout =  60000
06:28:01.017Z :client_base_impl              :318     Client 0 Test emiting  send_request
06:28:01.017Z :client_base_impl              :318     Client 0 Test emiting  send_chunk
06:28:01.024Z :client_base_impl              :318     Client 0 Test emiting  receive_chunk
06:28:01.025Z :message_builder               :134    Sequence Header SequenceHeader { sequenceNumber: 86, requestId: 36 }
06:28:01.025Z :message_builder               :257   client0 Sequence Number =  86
06:28:01.025Z :message_builder               :201   client0 message size = 66        body size   = 42        requestHandle =  36 ReadResponse
06:28:01.026Z :client_secure_channel_layer   :924   ------------------------------------- Client receiving response  ReadRequest 36 ReadResponse err= null securityTokenId= 1
06:28:01.026Z :client_base_impl              :318     Client 0 Test emiting  receive_response
06:28:01.026Z :client_session_keepalive_manag:144   emit keepalive
Session keepalive received.
```

mg90707, 2021-10-05:
I also made a series of dis- and reconnects with interesing results:

0. Initial connect to the PLC -> Status: good (subscription and monitored item registered)
1. Disconnect for ~20s -> Status: broken (no subscription and monitored items); client callbacks were _**not**_ invoked
2. Disconnect for ~20s -> Status: good; client callbacks were invoked
3. Disconnect for ~20s -> Status: broken; client callbacks were invoked
4. Disconnect for ~20s -> Status: good; client callbacks were invoked
5. Disconnect for ~40s (waited until "Client request timed out:" messages were logged) -> Status: broken; client callbacks were invoked

Here is the log output (only console.log statements this time without the DEBUG=ALL environment variable as that would get too verbose I think)
```
Subscription '1111923924' keepalive received.
Subscription '1111923924' keepalive received.
Client request timed out: { /*PublishRequest*/
 requestHeader                 /* RequestHeader       */: {
   authenticationToken         /* NodeId              */: ns=0;i=121045575
   timestamp                   /* DateTime            */: 2021-10-05T07:14:16.986Z
   requestHandle               /* UInt32              */: 9               0x9
   returnDiagnostics           /* UInt32              */: 0               0x0
   auditEntryId                /* UAString            */:
   timeoutHint                 /* UInt32              */: 40000               0x9c40
   additionalHeader            /* ExtensionObject     */: null
 }
 subscriptionAcknowledgements  /* SubscriptionAckno[] */: [ /* empty */ ]
};
xxxxx  <<<<<< _on_message_received for unknown or timeout request  9                PublishResponse Good (0x00000) 1124181364
Subscription '1111923924' keepalive received.
Subscription '1111923924' keepalive received.
Subscription '1111923924' keepalive received.
Subscription '1111923924' keepalive received.
Session restored.
Session keepalive received.
Client lost the connection.
07:15:45.309Z :client_session_keepalive_manag:124    warning : ClientSessionKeepAliveManager#ping_server  Connection Break
Session encountered a keepalive error: No state provided.
Client is starting the reconnection process.
Client finished the reconnection process.
dataValue =  { /* DataValue */
   value: Variant(Scalar<Null>, value: <null>)
   statusCode:      BadNodeIdUnknown (0x80340000)
   serverTimestamp: null
   sourceTimestamp: null
}
dataValue =  { /* DataValue */
   value: Variant(Scalar<Null>, value: <null>)
   statusCode:      BadNodeIdUnknown (0x80340000)
   serverTimestamp: null
   sourceTimestamp: null
}
dataValue =  { /* DataValue */
   value: Variant(Scalar<Null>, value: <null>)
   statusCode:      BadNodeIdUnknown (0x80340000)
   serverTimestamp: null
   sourceTimestamp: null
}
dataValue =  { /* DataValue */
   value: Variant(Scalar<Null>, value: <null>)
   statusCode:      BadNodeIdUnknown (0x80340000)
   serverTimestamp: null
   sourceTimestamp: null
}
dataValue =  { /* DataValue */
   value: Variant(Scalar<Null>, value: <null>)
   statusCode:      BadNodeIdUnknown (0x80340000)
   serverTimestamp: null
   sourceTimestamp: null
}
Session restored.
Subscription '1111923925' keepalive received.
Subscription '1111923925' keepalive received.
[... 133 lines cut ...]
 subscriptionAcknowledgements  /* SubscriptionAckno[] */: [ /* empty */ ]
};
Client request timed out: { /*PublishRequest*/
 requestHeader                 /* RequestHeader       */: {
   authenticationToken         /* NodeId              */: ns=0;i=2655200434
   timestamp                   /* DateTime            */: 2021-10-05T07:19:27.553Z
   requestHandle               /* UInt32              */: 19               0x13
   returnDiagnostics           /* UInt32              */: 0               0x0
   auditEntryId                /* UAString            */:
   timeoutHint                 /* UInt32              */: 50000               0xc350
   additionalHeader            /* ExtensionObject     */: null
 }
 subscriptionAcknowledgements  /* SubscriptionAckno[] */: [ /* empty */ ]
};
Client lost the connection.
07:20:20.508Z :client_session_keepalive_manag:124    warning : ClientSessionKeepAliveManager#ping_server  Connection Break
Session encountered a keepalive error: No state provided.
Client is starting the reconnection process.
Client finished the reconnection process.
Session restored.
Session keepalive received.
Session keepalive received.
Session keepalive received.
Session keepalive received.
```

mg90707, 2021-10-07:
Isn't this more a bug report than a support request? node-opcua fails to restore the session properly sometimes but thinks it did, even though it is detected that something is wrong. E.g.
`WARNING : SERVER TELLS THAT IT HAS NO SUBSCRIPTION , BUT CLIENT DISAGREE`
I mean in the end the server is the "source of truth" and the client has to react accordingly.

erossignon, 2022-02-27:
@mg90707 we need your support to work on bug

pro-xiaoy, 2022-03-11:
Hello, logs about **dataValue =**  At which step
thx

pa-olha, 2023-03-08:
Hi. My logs shows similar issue:
````
03:15:18.101Z :client_session_keepalive_manag:122    warning : ClientSessionKeepAliveManager#ping_server   serviceResult = BadSessionNotActivated (0x80270000)
03:15:38.448Z :client_session_impl           :143   client is now attempting to recreate a session
03:15:45.150Z :reconnection                  :252   Warning TransferSubscription has failed  serviceResult = BadServiceUnsupported (0x800b0000)
03:15:45.150Z :reconnection                  :253   May be the server is not supporting this feature
dataValue =  { /* DataValue */
   value: Variant(Scalar<Null>, value: <null>)
   statusCode:      BadNodeIdUnknown (0x80340000)
   serverTimestamp: null
   sourceTimestamp: null
}
dataValue =  { /* DataValue */
   value: Variant(Scalar<Null>, value: <null>)
   statusCode:      BadNodeIdUnknown (0x80340000)
   serverTimestamp: null
   sourceTimestamp: null
}
dataValue =  { /* DataValue */
   value: Variant(Scalar<Null>, value: <null>)
   statusCode:      BadNodeIdUnknown (0x80340000)
   serverTimestamp: null
   sourceTimestamp: null
}
dataValue =  { /* DataValue */
   value: Variant(Scalar<Null>, value: <null>)
   statusCode:      BadNodeIdUnknown (0x80340000)
   serverTimestamp: null
   sourceTimestamp: null
}
dataValue =  { /* DataValue */
   value: Variant(Scalar<Null>, value: <null>)
   statusCode:      BadNodeIdUnknown (0x80340000)
   serverTimestamp: null
   sourceTimestamp: null
}
03:15:46Z - [info] [opcua-client-connection] Session restored.
````

Last line of the log is written by event listener "session_restored". But in the code it is said: 

> session_restored is raised when the session and related subscription have been fully repaired after a reconnection
https://github.com/node-opcua/node-opcua/blob/master/packages/node-opcua-client/source/client_session.ts#L160

How comes the event is raised, while restoring the subscriptions failed?

erossignon, 2023-03-08:
I see two things her: 

 * the server doesn't support the TransferSubscription  service. This service is not always needed but forces the client to regenerate the subscription from scratch.
 * As the subscription and monitored item have been regenerated, it appears that the previously monitored item nodeids are not available anymore on the server side. 
 * this explain why BadNodeIdUnknown  is returned on the initial value statement
* The subscription repair process has been completed fully,, the subscription has not failed.  However the server doesn't expose the nodeid that used be monitored anymore, and there is nothing that the client can do about it.

The question to ask is why is the server is not exposing the node any more with the same node Id?

robertsLando, 2023-05-19:
I'm having this same issue too and it's happening when the OPCUA server is force closed (power turned on) and then node-opcua client isn't able to reconnect to it once it comes back

erossignon, 2023-11-13:
The culprit is that the server doesn't implement TransferSubscription

```
   TransferSubscription has failed  serviceResult = BadServiceUnsupported 
```

node-opcua needs to consider that possibility and fully recreate the subscription in this case.

robertsLando, 2023-11-13:
Yeah I think it's that

erossignon, 2023-11-13:
should be fixed now in [node-opcua@2.117.0](https://github.com/node-opcua/node-opcua/releases/tag/v2.117.0)

closed 2024-03-06 by erossignon