node-opcua/node-opcua #1072: Class UANamespace is not exposed using typescript

MarcelDe, 2021-11-02:
**Current behavior**

Class UANamespace is not exposed using typescript

**Describe the bug**

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

```import { UANamespace } from "node-opcua"``` fails because, ```"types": "./dist/source/index_current.d.ts"``` as defined in, ```package.json``` does not contain ```export * from "./namespace"```;

If ```import { UANamespace } from "node-opcua"``` is executed using JavaScript, it works fine.
This is because package.json specifies ```"main": "./dist/src/index_current.js"```. This file contains the export as expected.

Please pay special attention to the referred paths: ```src``` vs ```source```
 

**Expected behavior**
```import { UANamespace } from "node-opcua"``` to work using typescript.

As workaround to get the import working in typescript, the following line can be used:
```import { UANamespace } from "node-opcua-address-space/dist/src/index_current.js"``` 

**Screenshots**

**Gist**

 
gist:

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua as a package ( using npm install )

- **node-opcua version:** :
  "version": "2.50.0"

erossignon, 2021-11-03:
This is the intended behavior, UANamespace in an internal implementation class, 
please use Namespace instead.

erossignon, 2021-11-08:
To avoid further annoyance UANamespace has been fully deprecated and removed from node-ocpau@2.57.0 onward.
Namespace  has to be used instead , if needed

closed 2021-11-08 by erossignon