node-opcua/node-opcua #1074: Wrong format for array datavalue

lukesmolo, 2021-11-04:
**Current behavior**

**Describe the bug**
I declare an array as value for a variable, I read the variable and I get the entire datavalue for it. If I call the toJSON() method on the datavalue I get an object instead of an array as value of the variable. From UA-expert the value is correctly set as array. 

**Step by step instruction to reproduce**

Steps to reproduce the behavior:
This is my variable:
```
 {

            nodeId: 's=status',
            description: 'A description',
            browseName: 'Status',
            dataType: DataType.Int16,
            arrayDimensions: [5],
            value: {
                get: () => {
                    const value = [0, 1, 2, 3, 4                    ];
                    return new Variant({ dataType: DataType.Int16, arrayType: VariantArrayType.Array, value });
                }
            }
        }
```
and this is what I get from reading the variable:
```
DataValue {
  value: Variant {
    dataType: 4,
    arrayType: 1,
    value: Int16Array(5) [ 2, 1, 2, 2, 2 ],
    dimensions: null
  },
  statusCode: ConstantStatusCode {
    _value: 0,
    _description: 'The operation succeeded.',
    _name: 'Good'
  },
  sourceTimestamp: 2021-11-04T18:00:14.934Z {
    high_low: [ 30921125, -776151199 ],
    picoseconds: 409700000
  },
  sourcePicoseconds: 409783000,
  serverTimestamp: 2021-11-04T18:00:14.934Z {
    high_low: [ 30921125, -776149145 ],
    picoseconds: 615100000
  },
  serverPicoseconds: 615185000
}
```
Then if I call the toJSON() I obtain this:
```
{"value":{"dataType":"Int16","arrayType":"Array","value":{"0":2,"1":1,"2":2,"3":2,"4":2}},"statusCode":{"value":0},"sourceTimestamp":"2021-11-04T18:00:14.934Z","sourcePicoseconds":409783000,"serverTimestamp":"2021-11-04T18:00:14.934Z","serverPicoseconds":615185000}
```

**Expected behavior**
An array for value instead of an object where each key is the index of the expected array

**Context**

- (x) my request is related to node-opcua acting as an OPCUA CLIENT
- ( ) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- **node-opcua version:** :  2.55.0
  
  

- **Node:**
  `node --version` = 16.4.1

erossignon, 2021-11-05:
In fact, if you use the dataValue.toJSON file you will get a javascript object. 
The value property is of type Int32Array

I would recommend you to use the node-opcua-json package to convert the dataValue into a OPCUA Json compatible 

```
        const dataValue = new DataValue({
            statusCode: StatusCodes.BadAggregateListMismatch,
            serverTimestamp: new Date(Date.UTC(1789, 6, 14)),
            value: {
                dataType: DataType.Int16,
                arrayType: VariantArrayType.Array,
                value: [12, 13, 14]
            }
        });
        console.log(dataValue.toJSON());
/*
  DataValue
{
  value: {
    dataType: 'Int16',
    arrayType: 'Array',
    value: Int16Array(3) [ 12, 13, 14 ]
  },
  statusCode: { value: 2161377280 },
  sourcePicoseconds: 0,
  serverTimestamp: '1789-07-14T00:00:00.000Z',
  serverPicoseconds: 0
}
*/
        console.log(JSON.stringify(dataValue.toJSON()));
/**
{"value":{"dataType":"Int16","arrayType":"Array","value":{"0":12,"1":13,"2":14}},"statusCode":{"value":2161377280},"sourcePicoseconds":0,"serverTimestamp":"1789-07-14T00:00:00.000Z","serverPicoseconds":0}
**/

```    

 The problem comes from the conversion of typed array (Int16Array /UInt32Array) to JSON.

closed 2021-11-05 by erossignon

erossignon, 2021-11-05:
A possible workaround is to to use:

```javacsript
    console.log(JSON.stringify(dataValue, (k, v) => ArrayBuffer.isView(v) ? Array.from(v) : v));
````