node-opcua/node-opcua #1077: Run Client & Subscriptions Indefinitely

james-sequoia, 2021-11-05:
Hi,

I'm trying to create a session that never times out. I've added the "keepSessionAlive: true" to the client but how do I make sure that my subscriptions run indefinitely? I currently have it set up as this:
```
const client = OPCUAClient.create({
    keepSessionAlive:  true,
    endpointMustExist: false,
    keepSessionAlive: true,
});

await client.connect(endpointUrl);

const session = await client.createSession();

const subscription = await session.createSubscription2({
    requestedPublishingInterval: 1000,
    requestedMaxKeepAliveCount: 20,
    requestedLifetimeCount: 7 * 24 * 60 * 60 * 1000,
    maxNotificationsPerPublish: 1000,
    publishingEnabled: true,
    priority: 10
});
```
This should mean that the session will only timeout after a week (7 d * 24 h * 60 m * 60 s * 1000 ms) of no "back and forth" between server and client, correct?

Thank you.

erossignon, 2021-11-05:
I am not sure why you would need to have a infinite session .

    requestedPublishingInterval: 1000,
    requestedLifetimeCount: 7 * 24 * 60 * 60 * 1000,

means that the subscription would keep accumulating data changes on the server side for up to requestedLifetimeCount*requestedPublishingInterval milliseconds.

( 7 * 24 * 60 * 60 * 1000 ) * 1000  =  7000 days => 1000 weeks => ~20 years.

In practice, we never do this.

the requestedLifetimeCount parameter is used to keep the session available when there is no direct connection between the client and the server (like during a network outage or a intentional disconnect of the client that keep the subscription alive on the server for later recovery)

james-sequoia, 2021-11-05:
Ah okay. I misunderstood what the subscription parameters mean then.

Thanks for your help, it clears up a lot!

closed 2021-11-05 by james-sequoia