node-opcua/node-opcua #1078: const engine = server.engine - is undefined

pelis, 2021-11-07:
**Current behavior**
This issue some what relates to issue #930
I am creating a OPC-UA server:
```
...
server.initialize(function () {  
   serverEngine = server.engine;
   // server.engine is undefined after node-opcua release 2.26.0
```
when the above initialize function calls back, server.engine is set.
This is true for node-opcua releases prior to 2.26.0 (that's what I had before upgrading)

Now I moved to release 2.57.0
In the above shown initialize callback, server.**engine** is now undefined
To solve the problem, the post_initialize event is required
```
server.on("post_initialize", function () { 
  serverEngine = server.engine;
  // server.engine is now set
}
```

So this is just for who ever runs into that trap

**Step by step instruction to reproduce**

- ( x) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (x) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - (x) node-red
    - ( ) other : please specify

- Device: PC_____

    - ( ) Windows : version : 10______
    - ( ) Linux : version : ubuntu 18.04______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: ______
    - Version: ______
    - Manufacturer/Software vendor: ______
    - link to the manufacturer or product site: https://

- **node-opcua version:** :
  
  

- **Node:**
  `node --version` =

erossignon, 2021-11-08:
Hi @pelis,
I would recommend that you move from the callback pattern to the async/await pattern as demonstrated in the "node-opcua by example book". I know that you are an early adopter of node-ocpua and have created code in the early days where async/await was not available yet.  It is worth revisiting the creation of your server. 

```
const server = new OPCUAServer({});

await server.initialise();

const addressSpace = server.engine.addressSpace;
// .. do the post initialize stuff here 

await server.start();
````

erossignon, 2021-11-08:
You may also want to check the error in the initlize callback, 
```
 const server = new OPCUAServer({});
    server.initialize(function (err) {  
       if (err) {
        console.log("Something is wrong in initialization ', err);
        process.exit(1);
         return; 
       }
        console.log("initialize", server.engine);
        serverEngine = server.engine;
    });
```

pelis, 2021-11-09:
Thanks Etienne.

I am now checking the return err and in fact there is an error (below) (for some reason I have not tested 'err' before).
I have to look into this in more detail, it seams that there is something odd with the certificate.
However, I just retested with node-opcua@2.26.0, no error in this case (tested on windows 10 and ubuntu 18.04).

So thanks again and thanks for the great work you are doing.
Please feel free to close this issue.

just for completeness, this is the err message:
```
"TypeError: Cannot read property 'position' of undefined
at f (C:\\tlabs\\tlabs-nodes\\node_modules\\node-opcua-crypto\\dist\\source\\explore_private_key.js:7:27)
at explorePrivateKey (C:\\tlabs\\tlabs-nodes\\node_modules\\node-opcua-crypto\\dist\\source\\explore_private_key.js:79:28)
at publicKeyAndPrivateKeyMatches (C:\\tlabs\\tlabs-nodes\\node_modules\\node-opcua-crypto\\dist\\source\\public_private_match.js:8:59)
OPCUAServer.<anonymous> (C:\\tlabs\\tlabs-nodes\\node_modules\\node-opcua-client\\dist\\verify.js:109:68)
at Generator.next (<anonymous>)\n    at C:\\tlabs\\tlabs-nodes\\node_modules\\node-opcua-client\\dist\\verify.js:8:71
at new Promise (<anonymous>)\n    at C:\\tlabs\\tlabs-nodes\\node_modules\\node-opcua-client\\dist\\verify.js:4:12
OPCUAServer.performCertificateSanityCheck (C:\\tlabs\\tlabs-nodes\\node_modules\\node-opcua-client\\dist\\verify.js:104:12)
at OPCUAServer.<anonymous> (C:\\tlabs\\tlabs-nodes\\node_modules\\node-opcua-server\\dist\\base_server.js:165:69)
at Generator.next (<anonymous>)
at fulfilled (C:\\tlabs\\tlabs-nodes\\node_modules\\node-opcua-server\\dist\\base_server.js:5:58)
```

closed 2021-11-09 by erossignon

bruegth, 2022-02-11:
@pelis @erossignon Got same error while using ECC private keys, while seems that this can't be read. 

Only RSA Key will be read:
https://github.com/node-opcua/node-opcua-crypto/blob/master/source/explore_private_key.ts

I assume with this PR(https://github.com/node-opcua/node-opcua-crypto/pull/10) only ECC Certificate can be read and no Keys, so then all Server side usage of ECC Certificates together with ECC Keys are currently not supported or?