node-opcua/node-opcua #1079: Node OPCUA Server with a C# Client

IbrahimOezen, 2021-11-08:
So I have created a node.js Server from your tutorial: https://github.com/node-opcua/node-opcua/blob/v2.1.3/documentation/creating_a_server.md

Now I wanted to Implement a Client with a GUI (using c#) to connect to the server. How should I start? Are there any guides to work with? Is it even possible?

Thanks in advance!

erossignon, 2021-11-11:
Hi Ibrahim, 
If you need to write your front-end application in C#, you will probably need to use an OPCUA Client Library in C#  (some commercial version exists) 

Alternatively, you could choose to develop your front-end in HTML using React or any other framework.
You can get inspired by the basic HTML panel samples here (https://github.com/node-opcua/node-opcua-htmlpanel) 

Sterfive.com can also provide you with a  dedicated framework that facilitates the integration of node-opcua in web-based applications. ( send us a mail at contact@sterfive.com )

erossignon, 2021-12-14:
Hi Ibrahim, I m not sure I was able to help you, In any case, let close the issue, as there is no activity anymore.

closed 2021-12-14 by erossignon