node-opcua/node-opcua #1081: Function break from Node-opcua 2.56.0 -> 2.57.0

Graos, 2021-11-09:
Node-opcua (Server) :2.57.0

Register a call back when server is closed

this.server.on("session_closed", this.closeSessionEventCallback);
closeSessionEventCallback(session) {
....
}
initiate a close from Client to Node-opcua server
session object passed as function parameter to the registered callback function has the following wrong values:

session.aborted = true and session.channel is undefined.

The above flag session.aborted  is false and channel contains a valid object in node-opcua version 2.56.0 when connection is closed from client.

erossignon, 2021-11-09:
Correct. the behavior is now the intended one. 

session.aborted = true  & session.channel is undefined is the  new expected behavior.

Graos, 2021-11-10:
Thanks for the information.

closed 2021-11-10 by Graos