node-opcua/node-opcua #1084: ArgumentOptions arrayDimensions is being overridden to [0]

Filipecordeiro, 2021-11-15:
**Current behavior**

When adding a method in server mode,  the array dimensions of each input/output argument almost always defaults to [0] regardless of the values passed for `valueRank` and `arrayDimensions`. This results in some strange behaviors like not being able to use matrices or some clients recognizing the input/output as as array even though the `valueRank` is -1.

**Describe the bug**

I tracked the bug to function `constructHookArgument` in file `node-opcua/packages/node-opcua-service-call/source/imports.ts`. 
As you can see this condition seems to be incorrect:
````typescript

    // fix missing ArrayDimension (The value is an array with one dimension.)
    if (options.valueRank !== 1 || !options.arrayDimensions) {
        options.arrayDimensions = [0];
    }
````
If you pass a `valueRank` of 2 and an `arrayDimensions` of [2,2] it will change `arrayDimensions` to [0].
If you pass a `valueRank` of -1 and no `arrayDimensions` it will also change `arrayDimensions` to [0].

I've initially changed the condition to:
````typescript
    if (options.valueRank === 1 0 && !options.arrayDimensions) {
        options.arrayDimensions = [0];
    }
````
but there are some internal parts in the code that are counting on this to set the `arrayDimensions` value, so I changed it to a more compatible version:
````typescript
    if (options.valueRank > 0 && (!options.arrayDimensions || options.arrayDimensions.length === 0)) {
        options.arrayDimensions = new Array(options.valueRank).fill(0);
    }
````

With this change, all works as expected.

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

Add a method using an input argument as following:
````Typscript
    const argument: ArgumentOptions = {
        name: "InputA",
        dataType: DataType.String,
        valueRank: 2,
        arrayDimensions: [3, 3]
    };
````

**Expected behavior**

When adding a method to a server if we pass for example a `valueRank` of 2 and an `arrayDimensions` of [3,3] the `arrayDimensions` value should not be overridden to [0].

**Context**

- my request is related to node-opcua acting as an OPCUA SERVER

- I have installed node-opcua as a package ( using npm install )

- Device: PC
- OS version: Windows 10

- **node-opcua version:** :
  This issue appears to not be fixed as of today's most current version (2.59.0)

- **Node:**
  `node --version` = v12.15.0

erossignon, 2021-11-24:
fixed in node-opcua@2.60.0 and beyond, please confirm

erossignon, 2022-03-27:
@Filipecordeiro , let close now.

closed 2022-03-27 by erossignon