node-opcua/node-opcua #1085: How to set Endpoint Reconnection Max Retry

james-sequoia, 2021-11-15:
I need to run a js file through node indefinitely and need node-opcua to try to connect to the PLC indefinitely if it ever fails to connect. From what I can see, the max retry of reconnecting for when an endpoint can't be connected to is by default set to "-1" which is will retry indefinitely. Will it truly try to reconnect to an endpoint forever?

My code:
```
client = OPCUAClient.create({
    keepSessionAlive:  true,
    endpointMustExist: false,
});

client.on("backoff", async (retry, delay) => {
    console.log("Backoff ", retry, " next attempt in ", delay, "ms");
});

await client.connect(plc_ip);

session = await client.createSession();
console.log("session created");

subscription = await session.createSubscription2({
    requestedPublishingInterval: 1000,
    requestedMaxKeepAliveCount: 20,
    requestedLifetimeCount: 12 * 60 * 60, // (12 hours * 60 minutes * 60 seconds) * (requestedPublishingInterval of 1000ms)
    maxNotificationsPerPublish: 1000,
    publishingEnabled: true,
    priority: 10
});
```

Thank you.

erossignon, 2021-12-02:
> Will it truly try to reconnect to an endpoint forever?

Yes, once the first connection has been established, node-opcua client will keep trying to reconnect to the server in case of of network outage or lost of connection.

james-sequoia, 2021-12-02:
Thank you for the clarification!

closed 2021-12-02 by james-sequoia