node-opcua/node-opcua #1086: Method calls with enumeration types don't work as expected

marcodisarno, 2021-11-18:
**Current behavior**

When adding enumeration types as input arguments for method calls, they always fail:

**Describe the bug**

We get an BadInvalidArgument as a status code and the bound callback is not executed.

Reason seems to be that the client sends back an integer type, while NodeOPCUA server expects the enum datatype. I think, however, the client's behaviour is okay, since numerical values could easily be mapped to the enum type.

**Step by step instruction to reproduce**

Steps to reproduce the behavior:

1. Start an NodeOPCUA server
2. Connect a client (UAExpert, FreeOPCUA)
3. Go to Server -> RequestServerStateChange (or any other Method using enumeration type input parameters)
4. Call it

**Expected behavior**

Bound method should be called with chosen input parameter.

**Screenshots**

This is from UAExpert:
[image]

This is from FreeOPCUA:
[image]

```
08:51:40.400Z :argument_list                 :136    checking argDefDataType  
 ...                                                          nodeId              : ns=0;i=852
 ...                                                          nodeClass           : DataType (64)
 ...                                                          browseName          : ServerState
 ...                                                          displayName         : null ServerState
 ...                                                          description         : locale=null text=
 ...                                                          isAbstract          : false
 ...                                                          definitionName      : 
 ...                                                          binaryEncodingNodeId: <none>
 ...                                                          xmlEncodingNodeId   : <none>
 ...                                                          subtypeOfObj        : Enumeration
 ...                                                          references    :   length =2
 ...                                                               +-> ------- HasProperty (ns=0;i=46) ------> [ns=0;i=7612             ]EnumStrings                              EnumStrings
 ...                                                               +-> <------- HasSubtype (ns=0;i=45) ------- [ns=0;i=29               ]Enumeration                              Enumeration
 ...                                                          back_references     :   length =0 ( references held by other nodes involving this node)
 ...                                                 Definition                   :
 ...                                                                              :   { /*EnumDefinition*/
 ...                                                                              :    fields                        /* EnumField        [] */: [
 ...                                                                              :      { /*0*/
 ...                                                                              :        value                     /* Int64               */: 0,0
 ...                                                                              :        displayName               /* LocalizedText       */: locale=null text=null
 ...                                                                              :        description               /* LocalizedText       */: locale=null text=null
 ...                                                                              :        name                      /* UAString            */: Running
 ...                                                                              :      },
 ...                                                                              :      { /*1*/
 ...                                                                              :        value                     /* Int64               */: 0,1
 ...                                                 .... TRUNCATED .....
08:51:40.410Z :argument_list                 :137    checking argDataType
 ...                                                          nodeId              : ns=0;i=6
 ...                                                          nodeClass           : DataType (64)
 ...                                                          browseName          : Int32
 ...                                                          displayName         : null Int32
 ...                                                          description         : locale=null text=
 ...                                                          isAbstract          : false
 ...                                                          definitionName      : 
 ...                                                          binaryEncodingNodeId: <none>
 ...                                                          xmlEncodingNodeId   : <none>
 ...                                                          subtypeOfObj        : Integer
 ...                                                          references    :   length =1
 ...                                                               +-> <------- HasSubtype (ns=0;i=45) ------- [ns=0;i=27               ]Integer                                  Integer
 ...                                                          back_references     :   length =0 ( references held by other nodes involving this node)
08:51:40.418Z :argument_list                 :154    MADI argDataType.nodeId.value is not equal to argDefDataType.nodeId.value
08:51:40.419Z :argument_list                 :155    MADI   argDataType.nodeId.value  6
08:51:40.421Z :argument_list                 :156    MADI   argDefDataType.nodeId.value  852
08:51:40.422Z :argument_list                 :181    MADI isArgumentValid returned false!
```
_Remark: "MADI debugs were added by me._

**Gist**

 
gist:

**Context**

- ( ) my request is related to node-opcua acting as an OPCUA CLIENT
- (X) my request is related to node-opcua acting as an OPCUA SERVER

- ( ) I have installed node-opcua from source ( using git clone)
- (X) I have installed node-opcua as a package ( using npm install )
- ( ) I am using an application that uses node-opcua

    - ( ) node-red
    - ( ) other : please specify

- Device: PC
- OS version: Windows X
    - ( ) Windows : version : 19042.1348
    - ( ) Linux : version : ______
    - ( ) MacOs : version : ______
    - ( ) Raspbian: version : ______
    - ( ) Other : specify : ______

- Description of the other OPCUA system I am trying to connect to:

    - Name: UAExpert
    - Version: 1.5.1
    - Manufacturer/Software vendor: unifiedautomation
    - link to the manufacturer or product site: http://www.unifiedautomation.com/

    - Name: FreeOPCUA
    - Version: 0.8.0 
[image]
    - Manufacturer/Software vendor: FreeOPCUA Library
    - link to the manufacturer or product site: http://freeopcua.github.io/

- **node-opcua version:** :
  
  

- **Node:**
  `node --version` =

erossignon, 2021-11-19:
What is your node-opcua version?

marcodisarno, 2021-11-19:
latest - 2.59.0

Graos, 2021-11-22:
same as Writing to an Enum Value from external OPC UA client leads to error in node-opcua server bug Sponsors welcomed
#849

erossignon, 2021-11-24:
fixed in node-opcua@2.60.2

closed 2021-11-24 by erossignon