node-opcua/node-opcua #1088: Error on startup of an existing Server after update from 2.59.0 to 2.60.2

AndreasHeine, 2021-11-24:
**Expected behavior**
Starting without Error like in 2.59.0

**UMATI Demo Server**
change package.json node-opcua version to 2.60.2 (Renovate PR: https://github.com/AndreasHeine/SampleServer-node-opcua/pull/29)
https://github.com/AndreasHeine/SampleServer-node-opcua/tree/test

**Screenshots**

[image]

- **node-opcua version:** :
  2.60.2 

- **Node:**
  `node --version` = 14

```ts
    // add ExclusiveLimitAlarm
    const alarm = namespace.instantiateExclusiveLimitAlarm("ExclusiveLimitAlarmType", {
        browseName: "MySeverityCondition",
        conditionName: "MySeverityCondition",
        componentOf: dev,
        conditionSource: mySeverity,
        highHighLimit: 800.0,
        highLimit: 600.0,
        inputNode: mySeverity,
        lowLimit: 400.0,
        lowLowLimit: 200.0,
        optionals: [
            "ConfirmedState", 
            "Confirm",
        ]
    })
```

```
 LOG:   starting server... 
13:07:22.128Z :ua_variable_impl              :488   UAVariable   UAVariable.setValueFromSource HighHighLimit nodeId:ns=1;i=1027 dataType:ns=0;i=11:
 ...                                                the provided variant must have the expected dataType!
 ...                                                   - the expected dataType is Double
 ...                                                   - the actual dataType   is Int32
 ...                                                   - Variant(Scalar<Int32>, value: 800) HighHighLimit ns=1;i=1027
13:07:22.133Z :ua_variable_impl              :489   UAVariable.setValueFromSource HighHighLimit nodeId:ns=1;i=1027 dataType:ns=0;i=11:
 ...                                                the provided variant must have the expected dataType!
 ...                                                   - the expected dataType is Double
 ...                                                   - the actual dataType   is Int32
 ...                                                   - Variant(Scalar<Int32>, value: 800)
13:07:22.137Z :ua_variable_impl              :490   Error: UAVariable.setValueFromSource HighHighLimit nodeId:ns=1;i=1027 dataType:ns=0;i=11:
 ...                                                the provided variant must have the expected dataType!
 ...                                                   - the expected dataType is Double
 ...                                                   - the actual dataType   is Int32
 ...                                                   - Variant(Scalar<Int32>, value: 800)
 ...                                                    at UAVariableImpl.verifyVariantCompatibility (C:\Users\andre\Documents\GitHub\SampleServer-node-opcua\node_modules\node-opcua-address-space\dist\src\ua_variable_impl.js:484:23)
 ...                                                    at UAVariableImpl.setValueFromSource (C:\Users\andre\Documents\GitHub\SampleServer-node-opcua\node_modules\node-opcua-address-space\dist\src\ua_variable_impl.js:509:18)
 ...                                                    at UALimitAlarmImpl.setHighHighLimit (C:\Users\andre\Documents\GitHub\SampleServer-node-opcua\node_modules\node-opcua-address-space\dist\src\alarms_and_conditions\ua_limit_alarm_impl.js:149:28)
 ...                                                    at Function.instantiate (C:\Users\andre\Documents\GitHub\SampleServer-node-opcua\node_modules\node-opcua-address-space\dist\src\alarms_and_conditions\ua_limit_alarm_impl.js:77:23)
 ...                                                    at Function.instantiate (C:\Users\andre\Documents\GitHub\SampleServer-node-opcua\node_modules\node-opcua-address-space\dist\src\alarms_and_conditions\ua_exclusive_limit_alarm_impl.js:33:61)
 ...                                                    at NamespaceImpl.instantiateExclusiveLimitAlarm (C:\Users\andre\Documents\GitHub\SampleServer-node-opcua\node_modules\node-opcua-address-space\dist\src\namespace_impl.js:1267:74)
 ...                                                    at C:\Users\andre\Documents\GitHub\SampleServer-node-opcua\dst\serveraddressspace\serveraddressspace.js:101:29
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at C:\Users\andre\Documents\GitHub\SampleServer-node-opcua\dst\serveraddressspace\serveraddressspace.js:8:71
 ...                                                    at new Promise (<anonymous>)
13:07:22.151Z :ua_variable_impl              :521   UAVariable#setValueFromString Error :  HighHighLimit ns=1;i=1027
13:07:22.155Z :ua_variable_impl              :488   UAVariable   UAVariable.setValueFromSource 5:ChannelMode nodeId:ns=8;i=55233 dataType:ns=5;i=67:
 ...                                                the provided variant must have the expected dataType!
 ...                                                   - the expected dataType is Int32
 ...                                                   - the actual dataType   is UInt32
 ...                                                   - Variant(Scalar<UInt32>, value: 1) 5:ChannelMode ns=8;i=55233
13:07:22.159Z :ua_variable_impl              :489   UAVariable.setValueFromSource 5:ChannelMode nodeId:ns=8;i=55233 dataType:ns=5;i=67:
 ...                                                the provided variant must have the expected dataType!
 ...                                                   - the expected dataType is Int32
 ...                                                   - the actual dataType   is UInt32
 ...                                                   - Variant(Scalar<UInt32>, value: 1)
13:07:22.164Z :ua_variable_impl              :490   Error: UAVariable.setValueFromSource 5:ChannelMode nodeId:ns=8;i=55233 dataType:ns=5;i=67:
 ...                                                the provided variant must have the expected dataType!
 ...                                                   - the expected dataType is Int32
 ...                                                   - the actual dataType   is UInt32
 ...                                                   - Variant(Scalar<UInt32>, value: 1)
 ...                                                    at UAVariableImpl.verifyVariantCompatibility (C:\Users\andre\Documents\GitHub\SampleServer-node-opcua\node_modules\node-opcua-address-space\dist\src\ua_variable_impl.js:484:23)
 ...                                                    at UAVariableImpl.setValueFromSource (C:\Users\andre\Documents\GitHub\SampleServer-node-opcua\node_modules\node-opcua-address-space\dist\src\ua_variable_impl.js:509:18)
 ...                                                    at C:\Users\andre\Documents\GitHub\SampleServer-node-opcua\dst\machines\machinetool\showcasemachinetool.js:84:14
 ...                                                    at Generator.next (<anonymous>)
 ...                                                    at C:\Users\andre\Documents\GitHub\SampleServer-node-opcua\dst\machines\machinetool\showcasemachinetool.js:8:71
 ...                                                    at new Promise (<anonymous>)
 ...                                                    at __awaiter (C:\Users\andre\Documents\GitHub\SampleServer-node-opcua\dst\machines\machinetool\showcasemachinetool.js:4:12)
 ...                                                    at createShowCaseMachineTool (C:\Users\andre\Documents\GitHub\SampleServer-node-opcua\dst\machines\machinetool\showcasemachinetool.js:14:53)
 ...                                                    at C:\Users\andre\Documents\GitHub\SampleServer-node-opcua\dst\addressspace.js:23:61
 ...                                                    at Generator.next (<anonymous>)
13:07:22.184Z :ua_variable_impl              :521   UAVariable#setValueFromString Error :  5:ChannelMode ns=8;i=55233
 LOG:   error: Error:  creating addressspace failed: Error: UAVariable.setValueFromSource HighHighLimit nodeId:ns=1;i=1027 dataType:ns=0;i=11:
the provided variant must have the expected dataType!
   - the expected dataType is Double
   - the actual dataType   is Int32
   - Variant(Scalar<Int32>, value: 800) 
Waiting for the debugger to disconnect...
npm ERR! code ELIFECYCLE
npm ERR! errno 4294967295
npm ERR! sampleserver-node-opcua@1.0.0 start: `tsc && node dst/server.js`
npm ERR! Exit status 4294967295
npm ERR!
npm ERR! Failed at the sampleserver-node-opcua@1.0.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\andre\AppData\Roaming\npm-cache\_logs\2021-11-24T13_07_22_279Z-debug.log
Waiting for the debugger to disconnect...
The terminal process "C:\Windows\System32\cmd.exe /d /c npm run start" terminated with exit code: 4294967295.

Terminal will be reused by tasks, press any key to close it.
```

erossignon, 2021-11-24:
Yes; these errors are on purpose and show some types of mismatch error that was previously ignored.
I will need to investigate your code to help you fixing it.

closed 2021-11-24 by AndreasHeine

erossignon, 2021-11-24:
fixed in https://github.com/AndreasHeine/SampleServer-node-opcua/pull/31